python-microscopy/python-microscopy #323: [BUG]

csoeller, 2020-07-12:
**Describe the bug**

```
File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/pointcloud.py", line 263
SyntaxError: Non-ASCII character '\xe2' in file /Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/pointcloud.py on line 264, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

```

**To Reproduce**
Try to execute Mesh>Generate Isosurface in VisGUI

**Expected behavior**
No error.

More seriously, not sure why this gives a bug on my system but not on the developer's machines.

**Screenshots**

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.5
 - python-microscopy=20.07.10
 - python=2.7.17, platform=darwin
 - numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

**Additional context**
Quick test using

```
with open("pointcloud.py") as fp:
    for i, line in enumerate(fp):
        if "\xe2" in line:
            print i, repr(line)

```
gives

```
262 '            and A. K. Kenworthy, "On the use of Ripley\xe2\x80\x99s K-function and its \n'
264 '            pp. 1095\xe2\x80\x931103, 2009.\n'
```

Solution is presumably either

- delete offending nonascii chars

or

- `# coding: utf-8` at top of file?

zacsimile, 2020-07-12:
Sorry about that @csoeller. Happened on a copy/paste from Mendeley. Not sure why this ran on my machine. Fixed in #324.

David-Baddeley, 2020-07-12:
An explanation for it running in some places but not others would be py2 vs 3 - I think py3 defaults to UTF8 if no source encoding is declared, py2 possibly defaults to ASCII

David-Baddeley, 2020-07-12:
Merged now

David-Baddeley, 2020-07-12:
We need to find a way of automatically testing more of this stuff (or un-skipping more of the import unit tests which should have picked this up - we might now have got enough wx stuff out of the way that the import tests have a better chance of succeeding.

closed 2020-07-12 by David-Baddeley