python-microscopy/python-microscopy #325: [ENHANCEMENT]

csoeller, 2020-07-13:
**Is your feature request related to a problem? Please describe.**
Not specifically.

**Describe the solution you'd like**
It would be good to have a forum or similar place associated with PYME discussions (possibly just one forum or split user vs developer). Just having the issues list and random emails to individuals seems lacking IMHO given experience with previous open source projects.

Not sure if github offers anything or what is a good medium.

**Describe alternatives you've considered**
Maybe there is a good alternative I have missed. If there is already something I have missed it would be good to know.

**Additional context**
N/A

**Are you prepared to work on it?**
Happy to post silly questions and the occasional tip if I can come up with useful responses to questions posted.

barentine, 2020-07-13:
Hi @csoeller, I agree we and users would benefit quite a bit from a PYME forum! We're working to this end - have asked to be a community partner on image.sc ([here](https://forum.image.sc/t/pyme-would-like-to-be-a-community-partner/40187)). As I understand it we can go ahead and start using the forum (I suggest tagging posts with `pyme`) even before that goes through, and I appreciate your willingness to be involved there!

barentine, 2020-07-13:
Update that we are now community partners. New PYME posts there should use the tag `pyme` and one can view pyme-tagged topics [here](https://forum.image.sc/tag/pyme) where there is also the option to set up notifications.

David-Baddeley, 2020-07-14:
Andrew has chipped in already - as you can see this was in progress - with any luck the image.sc tag gets used more than our previous google groups forum. At this stage, I see image.sc mostly as fulfilling the user forum role, although that doesn't preclude development related discussions. We've also mooted having monthly (or thereabouts) developer zoom calls that you'd be welcome to join in on.

csoeller, 2020-07-14:
Great to hear that image.sc is there now. I have to admit, having "grown up" with mailing lists and associated archives, it feels a little different to have a broad forum that you narrow down with tags. I am probably just showing my age.

I would have thought it being nice to have a forum that is PYME only and controlled by PYME people, but David's comment about the google groups suggests that this was not really used while it existed.

Is there no better modern version of google groups that would be suitable? I have to admit I am probably out of touch with the latest tools.

David-Baddeley, 2020-07-14:
I'm also sightly wary, but happy to give it a shot. Between keeping the software up to date and moderation, running a forum these days is more or less a full time job, especially with troll farms and the like. Having someone else take care of the software and spreading the moderation responsibilities across a broader group of people sounds like a win. Hopefully we also generate a bit more exposure and traffic this way too.

csoeller, 2020-07-14:
Agreed. With that, happy to close this one...

closed 2020-07-14 by David-Baddeley