python-microscopy/python-microscopy #335: [ENHANCEMENT] allow dashes (maybe even spaces) in filenames to make cluster of one more user friendly

barentine, 2020-07-18:
**Is your feature request related to a problem? Please describe.**
I think people might be frustrated by our cluster filename restrictions
```
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/IO/unifiedIO.py", line 49, in assert_name_ok
    assert check_name(name) == True
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 263, in <lambda>
    self.bGo.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnGo(e))
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 657, in OnGo
    self.analysisController.pushImagesCluster(self.image)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 338, in pushImagesCluster
    metadata=self.resultsMdh, resultsFilename=resultsFilename)
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/HTTPRulePusher.py", line 191, in __init__
    clusterIO.put_file(resultsMDFilename, self.mdh.to_JSON().encode(), serverfilter=serverfilter)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/clusterIO.py", line 961, in put_file
    unifiedIO.assert_name_ok(filename)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/unifiedIO.py", line 51, in assert_name_ok
    raise AssertionError('Name "%s" is invalid. Names must only include alphanumeric characters and underscore' % name)
AssertionError: Name "/Users/Andrew/Documents/BewersdorfLab/nanoscopy-at-10k-data/analysis/4DNFISDEV89F.h5r.json" is invalid. Names must only include alphanumeric characters and underscore

```

**Describe the solution you'd like**
I'd like everyone, including myself, to just use nice directory / file names, but as you can see I myself got into trouble with dashes lately.

If I recall, we only use `~` as a special character, which I think is much less common in filenames/directory names than dashes. Maybe we could ease our filename restrictions a bit - even if only for cluster of one usage?

**Describe alternatives you've considered**
- leaving it be
- silently 'fix' the entire path and duplicate tree's on peoples hard drives for them ;)
- use a multi-character split in ruleserver to let us ease things up even beyond dashes. Current favorite is the hideous `~PyM3~`

**Are you prepared to work on it?**
yes, but preferably only a little

zacsimile, 2020-07-18:
I think allowing dashes and spaces is a good idea. Many of our users use both in their file-naming schema. 

At the very least, an improperly-formatted name should raise a `wx.Dialog` or something similarly user-friendly at the time of entry and not an assertion error after a run has already started.

David-Baddeley, 2020-07-19:
Just ran into this here on Friday too. At the very least we need to warn visibly and early about bad filenames when opened in dh5view, but somehow escaping them so they still work would be preferable. I think we should still try and enforce good filenames when saving data though. Spaces and hyphens in filenames are gross and have the potential to cause issues downstream when trying to use any command line utility to process the files.

I prefer the escaping bad filenames approach to using long multi-character separators. Looking a bit more closely I don't think we actually ever split on `'-'` so the check might be a little too zealous when reading files (although potentially appropriate for saving). We might be able to modify `check_name()` to support hyphens too, at least when reading files. Spaces will still need to be escaped, but we might be able to use the standard html escaping functions.

I'll have a look at this.

David-Baddeley, 2020-07-19:
Might have wrong on the hyphen - it's in the posix portable filenames character set.

https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap03.html#tag_03_282

Spaces are definitely a problem, though.

David-Baddeley, 2020-07-19:
just about to push a change to `check_name` to make it accept hyphens. `unifiedIO.fix_name()` should deal with spaces for us (replace with underscores).

barentine, 2020-07-20:
Forcing spaces to underscores might mirror trees of people's HDDs, e.g. /Users/Andrew Barentine/PYMEData/... -> /Users/Andrew_Barentine/PYMEData/.... Is that OK?

David-Baddeley, 2020-07-20:
I think it's fine for anything under PYMEData. It's not ideal above.  I'm somewhat surprised that you can have a space in your home directory name.

If we want to avoid too many special cluster of one cases and a whole pile of pain when someone tries to run things on a proper cluster we should stick to POSIX naming restrictions. We might end up needing to special case the PYMEData directory base and say we fix naming under this  (the cluster part of the URI) and accept what the path to PYMEData is. I'm not actually sure if this needs any new special casing, or if the current relative filename logic will handle it. If we only write to PYMEData then we're not going to spew stuff across various locations on the hard-drive.

barentine, 2020-07-20:
For what it's worth, I don't have a space in my home directory name, was just pontificating on hypotheticals / nervous that people might run analysis on things in weird places (maybe not even below PYMEData directory). I think you're right though that if the dataserver only sees below it's root path and we default that to PYMEData we are all set, and it's OK if we are strict with no-spaces below that.

Thanks for making the changes to accept hyphens

David-Baddeley, 2020-07-20:
As it stands, if you run analysis in a weird place, the results still go into PYMEData (unless you've specifically configured it otherwise).

barentine, 2020-07-23:
Going to close this for now - thanks all

closed 2020-07-23 by barentine