python-microscopy/python-microscopy #337: [ENHANCEMENT] Officially move to wx4

zacsimile, 2020-07-18:
**Is your feature request related to a problem? Please describe.**
We get a lot of `wx` deprecation warnings on `wxpython==4.0.4` (#82), plus some `C++` object deletion errors (#246). 

**Describe the solution you'd like**
I'd like to officially move to wx4. If we did, we could eliminate the warnings by upgrading our `wx` calls. Currently we use the deprecated calls because they are necessary to maintain compatibility with wx3 (#207). Moving to `wxpython==4.1` theoretically fixes the `C++` errors, although I can't test this because I'd need to update all of our deprecated calls (`4.1` finally removes the functions deprecated since wx3). 

**Describe alternatives you've considered**
The alternative is to continue supporting wx3.

**Additional context**
wx4, including `wx==4.1`, works on Python 2.

**Are you prepared to work on it?**
Yes.

David-Baddeley, 2020-07-19:
This is already on the roadmap and in the contributing notes. Moving to a default of wx4 in Sept, dropping support for wx3 in Jan 2021.

closed 2020-07-19 by David-Baddeley

zacsimile, 2020-07-19:
Ahh sorry, I missed the wx part in roadmap. I don't see it in `Contributing.rst`, which I thought was the contributing notes. Which file contains the contributing notes?

zacsimile, 2020-07-19:
Also, we should update

> Several bits of the GUI are still broken on wx4 (most notably anything which uses TraitsUI, but also some of the less used bits of our GUI code)

Traits is fixed (see https://github.com/python-microscopy/python-microscopy/issues/92#issuecomment-660369958).

David-Baddeley, 2020-07-19:
TraitsUI on wx4 is slowly getting better, but still a ways from being 'fixed' (the specific inspection issue mentioned in #92  is fixed, but this was one of a large number of small issues).

zacsimile, 2020-07-19:
Can you tell me what these issues are so I can look for them/fixes for them? I've been using wx4 for about a year and the only issues I've had that are traits-specific is the introspection issue (fixed) and #259, which is an issue only for specific versions of traits (e.g. it's an issue on `traits==7.0.0`, but not on `traits==7.0.1`).

David-Baddeley, 2020-07-19:
I don't have them all to mind, but what I recall (somewhat fuzzy):
- there are a bunch of warning messages (not in itself the end of the world)
- deletion problems (traitsUI objects not getting deleted properly with the enclosing window and giving dead object errors)
- some cosmetic rendering issues.
- a couple of traits editors don't actually work on wx4 (can't remember which, but set editor rings a bell). These are currently worked around by not using them.

zacsimile, 2020-07-19:
Good to know. If these are preventing us from moving to wx4 in September, it would be nice to document the tracebacks explicitly so we can verify fixes. I will keep an eye out.

David-Baddeley, 2020-07-20:
Nothing other than the introspection stuff I would regard as blocking for
making wx4 default in sept, but enough to justify keeping backwards
compatibility to wx3 for a bit afterwards.