python-microscopy/python-microscopy #339: [BUG]

csoeller, 2020-07-19:
**Describe the bug**

**Note**: suggested bugfix in PR #338.

```
 Error whilst running Processing>Plot &Profile	Ctrl-K
==============================================
python-microscopy=20.07.10
python=2.7.17, platform=darwin
numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/profilePlotting.py", line 47, in OnProfile
    if (self.do.selectionMode == self.do.SELECTION_SQUIGGLE):
AttributeError: 'DisplayOpts' object has no attribute 'SELECTION_SQUIGGLE'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open image in dh5view
2. Activate line drawing mode
3. Attempt to plot profile
4. See error

**Expected behavior**
no bug.

**Screenshots**

I have a screnshot to attach, alas how do you do it (probably missing the obvious)?

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: for details see error snapshot above
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
None.

David-Baddeley, 2020-07-19:
I believe this should now be fixed with your PR? correct?

csoeller, 2020-07-19:
Yes, should be good.

closed 2020-07-19 by csoeller