python-microscopy/python-microscopy #340: [BUG] ROIExtractNR fit results dtype incompatible with tabular dtype unnesting

barentine, 2020-07-20:
**Describe the bug**
ROIExtractNR fit result dtype is kind of weird, and breaks saving results (at least on py3)

printing the dtype, followed by each `n` in `tabular.unNestDtype` to show what we're hitting the error on (i.e. `<`)
```
[('tIndex', '<i4'), ('x', '<f4'), ('y', '<f4'), ('A', '<f4'), ('data', 'f4', (15, 15, 1)), ('sigma', 'f4', (15, 15, 1)), ('sp', [('Xg0', '<f4'), ('Yg0', '<f4'), ('Xr0', '<f4'), ('Yr0', '<f4')])]
pixel size: 115.0
('tIndex', '<i4')
('x', '<f4')
('y', '<f4')
('A', '<f4')
('data', '<f4', (15, 15, 1))
<
ERROR:PYME.recipes.base:Error in recipe module: <__main__.ExtractROIs object at 0x7fd191c729e8>
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "secret/pyme_series_script.py", line 237, in execute
    inputPositions=positions,
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 206, in apply_simple
    return self.apply(*args, **kwargs)[next(iter(self.outputs))]
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 185, in apply
    self.execute(namespace)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/measurement.py", line 233, in execute
    res = tabular.FitResultsSource(r, sort=False)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 235, in __init__
    self.setResults(fitResults, sort=sort)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 256, in setResults
    self._keys = unNestDtype(self.fitResults.dtype.descr)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 209, in unNestDtype
    unList += unNestDtype(n[1], parent + n[0] + '_')
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 209, in unNestDtype
    unList += unNestDtype(n[1], parent + n[0] + '_')
IndexError: string index out of range
Traceback (most recent call last):
  File "code/pyWIF/tests/pyme_series_script.py", line 279, in <module>
    recipe.execute()
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "secret/pyme_series_script.py", line 237, in execute
    inputPositions=positions, 
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 206, in apply_simple
    return self.apply(*args, **kwargs)[next(iter(self.outputs))]
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 185, in apply
    self.execute(namespace)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/measurement.py", line 233, in execute
    res = tabular.FitResultsSource(r, sort=False)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 235, in __init__
    self.setResults(fitResults, sort=sort)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 256, in setResults
    self._keys = unNestDtype(self.fitResults.dtype.descr)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 209, in unNestDtype
    unList += unNestDtype(n[1], parent + n[0] + '_')
  File "/Users/Andrew/code/python-microscopy/PYME/IO/tabular.py", line 209, in unNestDtype
    unList += unNestDtype(n[1], parent + n[0] + '_')
IndexError: string index out of range

```

**To Reproduce**
Steps to reproduce the behavior:
1. try to use ExtractROINR

**Expected behavior**
save roi's into a tabular

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: osx 10.15.5
 - Python version: 3.6
 - numpy version: 1.16

**Additional context**
This fitfactory could use some doc, but I assume it's preferable to using my own recipe module to crop localization ROIs from a series? Ran into this error while testing a short/sweet recipe module to run the FitPoints module using ROIExtractNR, but then pipe the 'data' into an ImageStack to put in the recipe namespace.

David-Baddeley, 2020-07-20:
The ROI extraction fit abuses the fit results datatype and produces data output which doesn't map super well into a tabular object (definitely not serializeable as .csv). It currently has a very specific use case and is not particularly useful for anything else. Given the unpublished nature of the use case, I suggest we move discussion off here for now.

That said, I suspect that we might be giving rise to this error, and we might be able to make it work with small modifications to `unNestDType`.

barentine, 2020-07-20:
Sounds good - I'll send you a message. Thanks David.

David-Baddeley, 2020-07-21:
hopefully fixed in 2348394922a07d34d9371d509f7a009b406b0629

barentine, 2020-07-22:
Works great, thanks @David-Baddeley!

closed 2020-07-22 by barentine