python-microscopy/python-microscopy #356: [BUG] dh5view localization live-view broken by new dsview plugin stuff

barentine, 2020-07-24:
**Describe the bug**
After hitting 'go' in dh5view

```
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 266, in <lambda>
    self.bGo.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnGo(e))
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 660, in OnGo
    self.analysisController.pushImagesCluster(self.image)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 348, in pushImagesCluster
    self.onImagesPushed.send(self)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/dispatch/dispatcher.py", line 162, in send
    response = receiver(signal=self, sender=sender, **named)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 683, in OnImagesPushed
    self.dsviewer.LoadModule('LMDisplay')
  File "/home/smeagol/code/python-microscopy/PYME/DSView/dsviewer.py", line 253, in LoadModule
    modules.loadModule(moduleName, self)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/__init__.py", line 118, in loadModule
    ret = mod.Plug(dsviewer)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMDisplay.py", line 324, in Plug
    return LMDisplay(dsviewer)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/LMDisplay.py", line 76, in __init__
    visCore.VisGUICore.__init__(self)
  File "/home/smeagol/code/python-microscopy/PYME/LMVis/visCore.py", line 88, in __init__
    gl_pan = wx.Panel(win)
TypeError: Panel(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 1 has unexpected type 'weakproxy'
```

**To Reproduce**
Steps to reproduce the behavior:
1. click 'Go' on a series in dh5view at the latest head

**Expected behavior**
open a live view

**Screenshots**
[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
linux
 - Python version:
3.6
 - numpy version:
1.16
 - wxpython version [if a UI bug]:
4

**Additional context**
Could alternatively just open PYMEVisualize with PR #347

zacsimile, 2020-07-28:
The `FRC` module in `DSView` is similarly broken. Googling about now.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/modules/coloc.py", line 389, in OnFRC
    dlg = ColocSettingsDialog(self.dsviewer, voxelsize[0], names, show_bins=False)
  File "/Users/zachcm/Code/python-microscopy/PYME/DSView/modules/coloc.py", line 33, in __init__
    wx.Dialog.__init__(self, parent, title='Colocalisation Settings')
TypeError: Dialog(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 1 has unexpected type 'weakproxy'
``

barentine, 2020-07-28:
Should be fixed with merge of #363 .

closed 2020-07-28 by barentine