python-microscopy/python-microscopy #367: [BUG] scalebar unit in dh5view looks like lm

barentine, 2020-07-28:
**Describe the bug**
[image]
liter-meters

**To Reproduce**
Steps to reproduce the behavior:
open a series in dh5view on windows, py3

**Expected behavior**
um

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] Win10
 - Python version: 3.6
 - numpy version: 1.16
 - wxpython version [if a UI bug]: 4.1.0

David-Baddeley, 2020-07-28:
works fine on OSX, wx3, py2.7.

expected behaviour is to show a greek mu.

Likely a unicode literal problem (or windows font problem - do the std windows fonts support all unicode chars?).

See `PYME.DSViewer.LMAnalysis` line 354

David-Baddeley, 2020-07-28:
If a windows font problem, could try substituting `'\u0B5'`

see https://www.fileformat.info/info/unicode/char/00b5/index.htm

David-Baddeley, 2020-07-28:
`'\u00B5'` confirmed to work on OSX (so safe substitution if it works on windows)

barentine, 2020-07-28:
Yep, that works.
[image]

Pretty sure this has been around for ages and is windows rather than py3/wx4, but I hardly use dh5view on windows, and I think people've just dealt with it.

Thanks David.

barentine, 2020-07-29:
fixed in ce54ff2

closed 2020-07-29 by barentine