python-microscopy/python-microscopy #381: [BUG] Splitter fits don't work with PYMEClusterofOne

David-Baddeley, 2020-08-03:
**Describe the bug**
Splitter fits do not work with PYMEClusterOfOne

**To Reproduce**
Run analysis on a ratiometric dataset

**Additional context**
This looks like  it is due to shiftmaps not being de-serialised from the .json metadata. I suspect this functionality was never written. Should be a reasonably simple fix, but blocking for py3 switch.

``` python
Traceback:
----------
Traceback (most recent call last):
  File "/Users/david/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "/Users/david/python-microscopy/PYME/localization/remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "/Users/david/python-microscopy/PYME/localization/FitFactories/SplitterFitScavNR.py", line 248, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "/Users/david/python-microscopy/PYME/localization/FitFactories/FFBase.py", line 169, in getSplitROIAtPoint
    DeltaX = self.metadata['chroma.dx'].ev(x_, y_)
AttributeError: 'unicode' object has no attribute 'ev'
```

barentine, 2020-08-03:
Would I be correct in guessing that this is using a shiftfield saved on python 2? Would you mind emailing me what I need to test, David?

David-Baddeley, 2020-08-03:
I think I'm pretty close to having it fixed (fingers crossed). Py2 in this case, but not the ground cause.

barentine, 2020-08-03:
Cool - I just remember running into (UnicodeDecodeError) hell trying to run the microscope in py3 and cluster in py2, and the compatibility flag not working. That said, I suspect support is better loading py2 in py3 than saving py2 compat in py3

David-Baddeley, 2020-08-03:
This is json, not pickle, so it shouldn't be an issue. Have a tentative fix which I need to merge back into the main branch.

David-Baddeley, 2020-08-05:
should be fixed as of 1038ea20ed44596b382118e20143de71ffecec3a

David-Baddeley, 2020-08-05:
Note - not yet tested on py3

barentine, 2020-09-15:
Tested on py3 windows cluster of one. I think we can close this. Thanks @David-Baddeley !

closed 2020-09-15 by David-Baddeley