python-microscopy/python-microscopy #389: [BUG] Unable to save isosurface meshes

lukasfue, 2020-08-04:
After generating an isosurface, I attempted to save the mesh and got this traceback:
 Error whilst running Mesh>Save mesh
==============================================
python-microscopy=20.07.29
python=3.6.10, platform=win32
numpy=1.14.6, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\extra_layers.py", line 316, in <lambda>
    visFr.AddMenuItem('Mesh', 'Save mesh', lambda e: save_surface(visFr))
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\extra_layers.py", line 131, in save_surface
    raise RuntimeError('No surfaces present')
RuntimeError: No surfaces present

Steps:
1. Load 3D localization data (from 4Pi)
2. Mesh --> Generate Isosurface --> Select <all> when asked with color channel --> Change N points min to 100; Remesh (on); Repair (on); Threshold Density set to 2e-05 --> OK
3. Mesh --> Save mesh --> Received traceback

Expected: The mesh would be saved as some file I could reload into visgui later on, obviating the need to redo the meshing which can take some time on larger datasets.

David-Baddeley, 2020-08-05:
Thanks @zacsimile. Can we close this?

lukasfue, 2020-08-05:
@David-Baddeley @zacsimile I just tested saving and loading meshes and it worked with no issues.

closed 2020-08-05 by lukasfue