python-microscopy/python-microscopy #392: [BUG] failing to remove intermediate directories for clusterofone io

barentine, 2020-08-04:
**Describe the bug**
If you specify a file on the command line, e.g. `dh5view PYMEData/4DN/4DNFISDEV89F.h5`, then `image.filename` gets set to that, and your nodeworkers can fail on io:
```
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 307, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 416, in __call__
    bufferManager.updateBuffers(md, self.dataSourceModule, self.bufferLen)
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 94, in updateBuffers
    self.dBuffer = buffers.dataBuffer(DataSource(md.dataSourceID, md.taskQueue), bufferLen)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/HDFDataSource.py", line 38, in __init__
    self.h5File = tables.open_file(self.h5Filename)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/tables/file.py", line 315, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/tables/file.py", line 778, in __init__
    self._g_new(filename, mode, **params)
  File "tables/hdf5extension.pyx", line 374, in tables.hdf5extension.File._g_new
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/tables/utils.py", line 154, in check_file_access
    raise IOError("``%s`` does not exist" % (filename,))
OSError: ``/home/ubuntu/PYMEData/PYMEData/4DN/4DNFISDEV89F.h5`` does not exist
```

**Expected behavior**
don't just add the entire PYMEData path if we're actually below it

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
ubuntu 18.04
 - Python version:
3.7
 - numpy version:
1.19
 - wxpython version [if a UI bug]:
4.0.4

barentine, 2020-08-04:
issue is directly using `image.filename` in `LMAnalysis`

```
self.pusher = HTTPRulePusher.HTTPRulePusher(dataSourceID=image.filename,
                                            metadata=self.resultsMdh, resultsFilename=resultsFilename)
```

David-Baddeley, 2020-08-04:
It's a little more complex as `image.filename` should be either fully
resolved ie /Users/... or relative to PYMEDataDir (i.e. should be safe to
use)

barentine, 2020-08-04:
right, sorry I didn't update that I sorted that out. Will submit PR shortly, thanks @David-Baddeley !

barentine, 2020-08-05:
fixed in #393 and #394

closed 2020-08-05 by barentine