python-microscopy/python-microscopy #402: [BUG] piezo_e816T gcs error results in NAN, locking up wx sliders

barentine, 2020-08-11:
**Describe the bug**
It's possible to hit a gcs error which throws hard and therefore doesn't get handled nicely by the current errcode check/log.

```
Exception in thread Thread-28:
Traceback (most recent call last):
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\Bergamot\Anaconda2\envs\pyme\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\Piezos\piezo_e816_dll.py", line 278, in _Loop
    self.position[0] = float(gcs.qPOS(self.id, b'A')[0])+ self.osen
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\GCS\gcs.py", line 102, in __call__
    self.HandleError(ID)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\GCS\gcs.py", line 14, in HandleError
    raise RuntimeError('GCS error: %d - %s' % (errno, TranslateError(errno)))
RuntimeError: GCS error: -1 - b'Error during com operation (could not be specified)'
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\focus_lock_gui.py", line 73, in refresh
    self._offset_slider.SetValue(int(100 * (offset - offset_range[0])))
ValueError: cannot convert float NaN to integer
Warning: not keeping up with camera, giving up with 0 frames still in buffer
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\focus_lock_gui.py", line 73, in refresh
    self._offset_slider.SetValue(int(100 * (offset - offset_range[0])))
ValueError: cannot convert float NaN to integer
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\focus_lock_gui.py", line 73, in refresh
    self._offset_slider.SetValue(int(100 * (offset - offset_range[0])))
```

We do use nan in `PYME.Acquire.Hardware.focus_locks.reflection_focus_lock.ReflectedLinePIDFocusLock`, however only internally to signal a failed fit. We check that the return is non nan and don't ask the offsetpiezoserver to change the offset if it is (and if we want to be extra sure, we can easily take that out and just return a success bool on the fit to toggle whether we change the offset, rather than checking success, setting nan, and then checking for nan).

Anyway, I'll add some logging and the next time it fails maybe get some more information, but I think gcs is giving us a nan the next time we call `float(gcs.qPOS(self.id,b'A')[iChannel]) + self.osen`, which we'll want to avoid setting `self.position[0]` with. My hunch is that doing so doesn't really break the piezo because it gets itself out on subsequent iterations of the loop, but the first time wx gets the position as nan and it makes its way into the offset range slider (or before we had that, to the normal position slider) we get stuck on nans and have to restart.

**To Reproduce**
Steps to reproduce the behavior:
1. lock
2. break the lock, e.g. with really weird multiwell chamber, or by setting the P gain really high or so

**Expected behavior**
log a statement saying we failed to get something from the com and we don't really know what the error is, but hold steady until things sort out rather than propagating nans into the gui

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] win10
 - Python version: 3.6
 - numpy version:  1.19
 - wxpython version [if a UI bug]: 4.1.0 (though I've had this error for quite some time, e.g. before I was using this version)

David-Baddeley, 2020-10-28:
From a cursory look it would appear that the slider should be re-set the next time around the loop / timer, and I'd expect this to resolve itself next time the timer ticks (if the piezo has recovered). I wonder why this isn't happening ...

barentine, 2020-10-29:
Sorry @David-Baddeley , should’ve mentioned I think we have the sliders down pat now - haven’t seen the nans in a bit. The GCS error code -1 still happens often on start up, particularly if the focus lock has thrown a lot of hard corrections rapidly and the pifoc silently (traceback silently, but audibly click) clicked off. I moved this in the project board for discussion but should’ve updated it / closed and opened a new one

barentine, 2020-10-30:
going to close this and update #515 as we have the gui slider bit solved but handling the e816T error code -1 in a way we got the piezo back up would make my life substantially easier

closed 2020-10-30 by barentine