python-microscopy/python-microscopy #409: [ENHANCEMENT] plugin installer

barentine, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
Discussed recently and I think raised by @zacsimile or @David-Baddeley:
As we get the gui-installers up to speed for more packages, it would be great if we could leverage it for plugins like nep-fitting, such that someone doesn't have to then find the installer env and manually conda or setuptools a plugin into that, which kind of defeats the purpose of the gui-installer.

**Describe the solution you'd like**
maybe at the top of `modules` in the main ui programs we have an `get plugins` menu item or so. There was some discussion about registering plugins via PR similar to `conda-forge/stage-recipes`, which I think would work quite well (and the requirement that plugins have conda recipes for gui-only install is a modest requirement).

**Are you willing to work on it?**
Yes, and I think `nep-fitting` would be a great test-case.

csoeller, 2020-08-15:
I agree, defeinitely a nice to have if it can be worked out how to do it robustly.

David-Baddeley, 2020-08-15:
As mentioned in the earlier conversation, I'm not overly enthusiastic about making plugin installation much easier than `conda install plugin-name` (or potentially `conda install -c author_name plugin-name`). I'm not completely dismissing the idea of having some kind of plugin installer, but do not consider it a priority as there are a bunch of difficult issues to address. Anything that we can do to a) increase plugin visibility and b) make it easy to find and install plugins is however good.

The reasons are as follows:
- Plugins are arbitrary executable code, and have significant security implications. If we offer plugin installation from a menu I believe we have a responsibility to perform some form of audit/curation of the plugins that are installable that way. We should ideally also verify that they work with the current version to avoid user frustration. This is an enormous amount of work which we do not have the resources for. 
- We don't really want to be in the business of providing support for 3rd party plugins, so it needs to be clear that plugins are separate entities with separate maintainers.

To be acceptable, any installation method can't be too simple - it has to be clear that this is not part of the core PYME and that you should make your own decisions about whether you want to trust the code. The current method of `conda install -c author_name plugin-name`, in my mind, ticks all the boxes - it is not particularly onerous, but makes it a conscious decision to install someone else's code.  

This is not to say that we shouldn't do more to make it easy to find and install plugins. The middle ground that was part of the original discussion was to either:
1) maintain a list on the website and/or in the help of plugins with name, author, and `conda install ...` command, encouraging people to PR to this list when they create a plugin.
2) create a dedicated conda channel for plugins and a repository (not unlike conda-forge) where people could submit conda recipes for their packages. This would make it easy to ensure that plugins were built for all plugins on the same versions of python and numpy that core PYME packages are available for, and provide a common place for people to discover plugins.

Combined with suitable documentation, I suspect this would be good enough without running into many of the headaches that running a plugin 'app-store' would entail.

zacsimile, 2020-08-15:
I agree that we don't want to run an app store. What is the key difference between option 1 and a window that lists these same options with an "Install" button next to them? Would a dialog alert that pops up when a user presses "Install" that says "we take no responsibility for this code, please make sure you trust it before installing" be sufficient?

David-Baddeley, 2020-08-15:
Users are conditioned to click buttons - from a security perspective it's better to require a more deliberate action (e.g. typing something). This could potentially happen through the GUI (see comment on Andrew's PR), but I do wonder if doing so is not just a lot of effort for something which is not intrinsically broken (maintaining a simple list is a lot easier than maintaining a list and a corresponding GUI).

barentine, 2020-08-15:
To be acceptable, any installation method can't be too simple

Seriously?  We want this to be broadly used and not let people install things easily? What's the point of the exe installer if `conda -c stuff something` is good enough? We want to split pyme chunks out into plugins and not make it easy to install them? None of this makes sense to me. Security concerns are pretty straightforward - we only have the plugin list pull options from our conda-forge style staging, call it `python-microscopy/plugin-staging`. You won't have to do any vetting yourself unless you won't let others merge PRs. I'll do it all, even, I don't suspect we're about to run into wild amounts of overhead here. Finally, I'll quote the bsd-2 clause we suggest in the plugin template repo:

> IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

David-Baddeley, 2020-08-16:
You are quoting that initial line out of context ... 

For the most part `conda install ...` *is* good enough. For me the executable installers solve two specific problems:

1) conda keeps changing what is in their channels, and what versions of python etc ... are bundled with miniconda/anaconda. Because of our large number of dependencies, this breaks the install process regularly. The executable installers let us effectively freeze all of this in a known working configuration. This is much less of an issue with plugins which are expected to have a much smaller dependency footprint (and our executable installers should solve this issue for plugins too).
2) (and much less importantly) initial adoption threshold. conda-installing is a little out of the comfort zone of many biological users, meaning that they need to be fairly motivated to try it. Again, I see this as less of an issue with plugins as users will hopefully have some buy in to the software already from installing pyme* (and plugin-installation is then only one additional step rather than multiple steps for a full PYME installation).

The license that plugins choose to use is completely irrelevant to this discussion -  it is a contract between plugin-authors and users and has nothing to do with us. My concern is more ethical than legal, but seeing as you bring it up I will note that you cannot contract out of much of the legislation (GDPR comes to mind, but there is a bunch of other stuff too) that we would potentially infringe on if we let a malicious plugin through so there is the potential, albeit small, for legal liability here. 

At present, with nep-fitting being the only widely available plugin, the maintenance overhead would be small, but if we are serious about growing our user and contributor base we would hope that this doesn't remain so. In growing the community I also think that it's important to offer consistency to plugin authors - conda installing plugins is something that I'm prepared to commit to always supporting, but it would be super frustrating if we were to have to revoke installation via a menu at some point in the future when moderation was becoming too onerous. Moderation/vetting is also a barrier to people wanting to write plugins.

None of this precludes us from bundling commonly used (and trusted) plugins (e.g. nep-fitting) with the executable installer - breaking things down into plugins etc ... rather than putting it in all in the core is about maintenance and ownership more than anything else. It doesn't imply that and end user should necessarily have to install everything separately.

*[this neglects plugins as a potential entry route to PYME, but that can likely be addressed by bundling some of the cooler plugins with the installer]

barentine, 2020-08-16:
I disagree that conda install is good enough. We know command line installation (simple or not) is a blocker for many people, we know other successful tools with plugins [allow people to install them graphically](https://imagej.github.io/list-of-update-sites/), and we are considering whether to intentionally not make this as simple as possible.

Anaconda Inc. and conda-forge let people upload/install whatever.  How specifically could GDPR and a bunch of other stuff make us liable for allowing people to install software they are already able to install? Is it different legally to conda install via gui than conda install via command line? Being that you can conda install through a gui in anaconda navigator...?

Ethically, I think selecting 'package' and clicking 'install' is not different re:understanding risk to typing `conda install package`. It would be nice if we have a look at plugins before merging them into the staging area. We can handle vetting. If it becomes onerous, I would be incredibly happy at our success, and further it would indicate there is a large developer community that might even be happy to share the burden.

barentine, 2020-08-19:
The other day I opened an ipynb in vscode with a conda environment that didn't have `ipython-kernel`, a dialog in the lower right popped up and said 'you don't have this, would you like to install it?' and clicking yes opened a terminal, activated the conda env, and ran `conda install -y ipython-kernel` in the terminal. It was pretty slick, and I wonder if that might also be a compromise here. We could consider having a `plugins` submenu in `modules` which lists all of the plugins and if you try to load one you don't have installed you are then prompted? That plugin menu could get pretty populated - I don't know.

David-Baddeley, 2020-08-19:
I know I'm a bit too conservative about this, but having lived through the Internet Explorer and activex days as well as, e.g. outlook and acrobat bugs I don't want to create the next malware vector. This might be partly aspirations of grandeur as to how broad our adoption could be. I think that you can get away with a little more with something that targets developers (VSCode, conda-forge, etc) as they should be aware of potential issues. The `conda install -y ipython-kernel` example above is also a bit different in that it's installing a known trusted package from an implicitly trusted channel (either the default channels or something that the user has explicitly added).

zacsimile, 2020-08-20:
Can we try maintaining a list of trusted plugins as a file in the main repo (option 1) for a bit and see how that feels? As long as this list is from folks we trust, I think eventually it should be OK to have the conda-uploaded ones install via the GUI. I propose we have a `plugins.yaml` that includes the package name, author, description, repo address, and conda channel (if available) for installation.