python-microscopy/python-microscopy #411: [ENHANCEMENT] tileviewer usability improvements

barentine, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
As we get towards making figures from overviews it would be nice if the viewer:
- could save the render to png
- had a scalebar overlay
- could load ROI locations from file
- set ROI size (I wrote this previously in a PR, but we took it out for some reason)

**Are you prepared to work on it?**
surely, though if I do I don't want it hung up again (honestly can't remember why the roi size got kicked back out - maybe because it was in units of pixels rather than um and if you change the microscope active camera, etc., etc.)

David-Baddeley, 2020-08-18:
Some of these have the potential to be pretty hard: 
- saving to PNG is going to be really hard [the viewer is like google maps and never actually creates one image with all tiles in it - each tile is rendered separately to the screen]. I think the best option here is to just take a screenshot.
- loading ROI locations from file might also be hard (due to the fact that the web-browser can't access the local disk). If we want a nice GUI we'd need to basically recreate a file -> open dialog from scratch with an appropriate backend in the server. Pragmatic solution would likely be to pop up a text field for the user to enter a full path to a file into which was then loaded on the server side, but this is a) gross in terms of UI and b) problematic security wise if you are ever running it on anything other than localhost.

The ROI size got stalled on some implementation issues, but is definitely still on the TODO list. I think it initially didn't make it through because of the difficulty of keeping it synchronised with the current ROI size in PYMEAcquire, rather than a units issue (remembering that the tileviewer is not HTSMS specific and we don't have fixed ROIs which would lend themselves to being specified as a command line argument to the tileviewer in most of the other microscopes). There was also an usability /aesthetic concern with the idea of having it as a text field in the GUI and the possibility of this being mistaken for being able to set the size of the ROI which would be acquired. I think there are two more or less reasonable solutions - make it entirely client side and have it in an expandable "settings" panel (could also include scale bar size, visibility, etc) which is hidden by default, or wait until we have the acquire server up and running at which point the true ROI size should be easily accessible.

Scale bar overlay shouldn't be too hard (fingers crossed).

David-Baddeley, 2020-08-18:
On second thoughts, loading ROI locations might be able to be addressed with a file upload with less work than completely recreating a file browser, but would still require a bit of messing around in javascript.

David-Baddeley, 2020-08-18:
Part of me is also unsure how much effort we should invest in making the tileviewer perfect as the multi-resolution web based image viewer / dh5view replacement which is currently in the roadmap will hopefully end up replacing this (although this could be quite some ways off). I think we should do what is definitely blocking (uploading locations seems like it might fall into this category), but should also think about what is necessary vs nice to have.