python-microscopy/python-microscopy #415: [ENHANCEMENT] io plugins for pipeline construction

csoeller, 2020-08-15:
**Is your feature request related to a problem? Please describe.**
There are different file formats out there for event storage and I was hoping it be feasible to come up with an 'import' plugin infrastructure. This is also along the lines of "becoming the ImageJ of SMLM viewers".

**Describe the solution you'd like**
Similar to the way plugins for VisGUI already exist an io import structure would be useful (e.g. for csv with particular column name mapping; picasso hdf5; etc). The plugin could register an import function that provides the ingredients to construct a pipeline + metadata. The user could explicitly use that import function from the menu. No fancy auto-detection required but could be considered if not too much 'opening a can of worms'.

**Describe alternatives you've considered**
One can convert, e.g. I have a converter for picasso to pyme friendly CSV but would really like a visgui IO plugin API.

**Are you prepared to work on it?**
Once we settle on a way how it would and should work, sure.

David-Baddeley, 2020-08-16:
Definitely a worthwhile objective. Can think of a few wrinkles, particularly when it comes to metadata and name translation etc ...  Something to keep in mind when we incorporate #284 (or a modified variant thereof).

Is likely going to involve:

- refactoring `pipeline._ds_from_file()`
- writing a new `tabular` source for the desired file type.

Initially, binding it through something like `File->Import` in VisGUI would probably be desirable (lets the user choose the filetype rather than having to do detection magic), but ultimately some form of detection stuff would also be good.

Have you got any other formats in addition to picasso in mind? Where are they documented?

Unclear at this stage if they should be plugins, or just have things refactored and documented so that it's easy for someone to contribute a loader without wading through the complicated historical mess that is the current data import process. I'd imagine most loaders would only be a single file or so.

csoeller, 2020-08-16:
_Have you got any other formats in addition to picasso in mind? Where are they documented?_

It seems like Thunderstorm at least has a set of known CSV columns that should be considered, though not sure if a proper format description (see this post [here](http://www.neurocytolab.org/tscolumns/)).

This general idea also relates to another thought I had and will get started on RSN, i.e. starting a repo of SMLM example data, using our own and any available external data that is interesting. It would serve as a convenient set of test data to validate IO against, try some methods with, see if old PYME formats can still be properly digested, etc. It's something that has bugged me for a while, always having to search for good example data in the depth of my file system or elsewhere.

I will attempt to collect format description links if I find any, most formats seem adhoc or undocumented these days. Also not sure what Zeiss, Nikon, etc use, probably internal stuff but it clearly allows CSV type exports.

Anyway, I would be very happy to see an infrastructure that allows testing of new IO formats that can then be considered for proper integration. I'd be happy if I could come up with a plugin 'hack' on top of `pipeline._ds_from_file()`  refactoring so that I can experiment with this in a _not-for-public-use_ plugin dist.

As you say, metadata could be a headache, possibly a companion file (like the `.md` file type) that gets manually created could be used to populate such info or and also make `csv`s straight importable. That could indeed be the simplest first step, a `.md` type rule that is searched for when a csv is openend so that no interaction is necessary and could be written after a manual CSV import for future openings without any user interaction. This could include a column mapping to PYME columns possibly.

David-Baddeley, 2020-08-16:
A set of example data has also been on my TODO list.

zacsimile, 2020-08-20:
Do we need to refactor these things? We already allow arbitrary column mapping for CSV and MAT files. We could just parse a `custom_input_mapping.yaml` if present. e.g. `pymevis my_file.csv -m custom_input_mapping.yaml` and pass the column names to `TextfileSource`. This puts the onus on the user to tell us what kind of file is present, but allows for reusable mappings. We could even store these reusable mapping in a `mappings` folder similar to our `PYMEAcquire` `Scripts` folder and allow users to PR custom mappings.

David-Baddeley, 2020-08-23:
Just as a quick note, the `.md` format is effectively deprecated in favour of `.json`, and in general we are moving more towards standard formats (e.g. .json, .yaml) in general.

I tend to lean towards actually factoring this out properly rather than trying to hack it into the current opening architecture. @zacsimile's .mat extensions were hard enough to do within the current framework. I'm also a bit nervous about the concept of dropping a metadata file in the same directory (regardless of format). It's not super intuitive and hasn't worked particularly well with dh5view.

csoeller, 2020-10-01:
Hi all, just thinking at some stage we should review how best to implement some kind of import for known external formats (not many I know of as yet, say picasso + ThunderSTORM csv). Not unexpectedly, with the interest that PYME got with the preprint people quickly want to open their files and making it as easy as possible is a big factor in adoption I would suspect.

There are obviously priorities to consider, would be nice to come up with a plan at some stage though, but I imagine it still would require refactoring of some current code first, where I could only make a limited contrib.

csoeller, 2021-05-27:
Hi all, I would like to bump this up for consideration again. With the PYME Nat Meth paper out (yeih!) I would expect people to be keen to have something like an import option for regularly encountered formats that does not need a complex import dialog.

Case in point, we are trying to import event data exported from Zeiss Elyra. Example below via a couple of screenshots. Some regular tasks seem to come up:

- there are names for columns that need mapping from their naming to our naming
- potentially scaling/unit conversions (possibly with info from metadata)
- generate new columns from some of the input columns (e.g. from halfwidth to sigma, photon numbers etc)
- some kind of filtering or similar to ignore problematic file parts (see strange mostly empty rows at end of Elyra export)

Eventually it may be good to come up with a strategy for that (and possibly also make these pluggable) but as David has mentioned this may need some internal refactoring first. If somebody has an idea how to get something quickly, e.g. a converter, or a dirty but working hack for the interim...

**Start of Elyra exported file**

[image]

**End of Elyra exported file**

Note the garbage lines that trip up our standard importer and currently need manual deletion.

[image]

David-Baddeley, 2021-05-27:
A few quick thoughts:

- it probably makes sense to handle different .csv flavours as sub-cases of the current text importer rather than with separate importers (to save duplication, but also because there is not necessarily a simple way (e.g. by extension) of unambiguously recognising a 'flavour'). I think there still needs to be a chance to modify the flavour manually in the text import dialog.
- To import the Elyra data using the current importer we'd need:
    - a better way of recognizing and/or skipping header lines (rather than relying on them having leading comments)
    - a way of ignoring the lines at the end. This could probably be accomplished by setting the comment character to `,`  / allowing the user to set a custom comment character.
- On top of that, it would be nice to have (in order of decreasing priority)
    - a map of preset column-name translations for each .csv flavour
    - a way of automatically detecting a given flavour
    - a series of data-mappings/transformations to apply.

My gut feeling is that most of this could be accomplished using a dictionary (or a .yaml/json)  specification along the following lines:

```python
csv_flavours = {
'elyra' : {
    'regex' : r'pattern', # a regex which should match the first line of the file, used for detection
     'header_rows' : 1, # the number of header rows to skip before the first line of data
      'column_names_row' : 0, # index of the line which contains the column names
      'comment_char' : ',', # ignore lines starting with this character
      'column_name_mappings' : {  # a dictionary of name substitutions
           'in_name1' : 'out_name1',
            },
      'column_translations' : {}, # a dict to populate a MappingFilter (To be confirmed)
      },
}
```

Everything except for the column translations should be fairly easy to add, so I'd suggest punting the latter. For now I'd probably suggest just putting this in a python module within PYME - I think it's too lightweight for separate plugins, and probably lowest overhead to just have people pr a dictionary addition.

David-Baddeley, 2021-05-27:
I expect the flavour detection might not be particularly robust - at least initially so, we'd still display the full text input dialog, with a dropdown for the flavour, using the regex to try and pre-select the flavour automatically.

csoeller, 2021-05-27:
The `csv_flavours` approach seems a great way to get this started and it will become clear soon enough if there were important gaps.

To be honest, I was not so much thinking about flavour autodetection, indeed this could be tricky. I was just thinking an import menu with a bunch of choices could be one, good enough, way to implement this. After all, users should have a pretty good idea where their data comes from (Zeiss, Nikon, XXX, RapidSTORM) and just select the respective style in an import drop down like fiji, see below.

I would suggest the exact way how to fall back to a generic import interface in case we cannot autodetect or get an error on import can be experimented with in a second stage. Initially just having an if only clunky way to put the import flavours through their paces would be my choice.

[image]

David-Baddeley, 2021-05-27:
The thousands of menu items approach is one of the things that bugs me
about imagej. There is a certain pragmatism to it, but it's not what would
typically be considered good ui.

zacsimile, 2023-07-05:
Hi all. Bumping this again as I have a collaborator with ViSP files (`.3d` and `.3dlp`) that I want to open with PYME. Since these are ASCII tabular (see spec from the ViSP manual below) I currently convert them to `.txt` and then import them. It feels like I could natively support `.3d` and `.3dlp` in a similar manner to the CSV flavors. I think all we need to do is accept different extensions. Are there any objections to adding such behavior into the loading setup?

[image]

David-Baddeley, 2023-07-05:
I don't see why you shouldn't be able to import visp files directly using csv flavours (we might need to modify the list of extensions we recognise as text, add extension as a factor in auto-choosing a flavour and add a visp flavour(s)). 

It should definitely be an extension of csv_flavours rather than something standalone.

David-Baddeley, 2023-07-05:
Actually, you shouldn't need to convert to `.txt` as PYMEVis already treats anything that isn't `.h5r`, `.h5`, or `.mat` as text. IE `PYMEVis somefile.3d` should already work, albeit with manual column assignment. Basically all that is needed is adding extension as a selection criteria to csv_flavours, and adding the flavours for .3d and .3dlp.

David-Baddeley, 2023-07-05:
If you want to be able to use File->Open from the menu, you'll also need to modify `OnOpenFile()` to either specifically add `.3d` and `.3dlp` or just add a generic `All files` option with a *.* selector.