python-microscopy/python-microscopy #436: [BUG] race condition in get_advertised_services occasionally fails a couple localization tasks on start

barentine, 2020-09-01:
**Describe the bug**
If you queue up some localization tasks, often a couple 2-3 early frames will fail with 
```
Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 299, in locate_file
    locs, t = _locateCache[cache_key]
KeyError: 'villains::Bergamot/2020_8_21/PK208-059_nup96gfprbCF568/000/analysis/21_8_series_A_1.h5r.json'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/misc/sqlite_ns.py", line 103, in getNS
    ns = nsd[(protocol, thread_id)]
KeyError: ('_pyme-http', 140515308488448)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 307, in computeLoop
    task = remFitBuf.createFitTaskFromTaskDef(taskDescr)
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 292, in createFitTaskFromTaskDef
    mdh.update(json.loads(unifiedIO.read(md)))
  File "/home/ubuntu/code/python-microscopy/PYME/IO/unifiedIO.py", line 226, in read
    s = clusterIO.get_file(sequenceName, clusterfilter)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 792, in get_file
    locs = locate_file(filename, serverfilter, return_first_hit=True)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 314, in locate_file
    services = get_ns().get_advertised_services()
  File "/home/ubuntu/code/python-microscopy/PYME/misc/hybrid_ns.py", line 31, in get_advertised_services
    services = dict(self._sqlite_ns.get_advertised_services())
  File "/home/ubuntu/code/python-microscopy/PYME/misc/hybrid_ns.py", line 16, in _sqlite_ns
    return sqlite_ns.getNS(self._protocol)
  File "/home/ubuntu/code/python-microscopy/PYME/misc/sqlite_ns.py", line 105, in getNS
    ns = SQLiteNS(protocol)
  File "/home/ubuntu/code/python-microscopy/PYME/misc/sqlite_ns.py", line 41, in __init__
    self._conn.execute("CREATE TABLE dns (name TEXT, address TEXT, port INTEGER, creation_time FLOAT, URI TEXT)")
sqlite3.OperationalError: table dns already exists

```

**To Reproduce**
Steps to reproduce the behavior:
1. Start the cluster
2. pick some series
3. `Localize molecules`

**Additional context**
I forgot to configure a node to ditch the hybrid ns in clusterIO, so that's an easy fix, but regardless others might hit this, so we should probably either fix the race condition and/or consider switching the default. What is the use case for hybrid ns in clusterIO?

David-Baddeley, 2020-09-14:
The hybrid ns use case is a workaround for the fact that zeroconf / ClusterOfOne is super problematic on a crowded network (especially on Win10). So many things respond to and issue mdns queries that you often time out resolving the different servers, so falling back to sqlite based resolution is needed to make the analysis robust unless you pull the network cable on the computer. 

There is still a performance issue in that even with hybridns the zeroconf code still runs and the zeroconf thread spends a significant amount of CPU time just servicing the spurious traffic. Medium term this can probably be solved by running the zeroconf protocol on a non-standard port - see https://github.com/jstasiak/python-zeroconf/issues/186, but the hybridNS stuff is probably still useful as a backup. 

The performance issue could also be solved by making the sqlite based resolution to be default, with zeroconf strictly opt-in (when using on a cluster where we control the network and can eliminate spurious traffic). 

In any case, we should definitely add the locking needed to avoid this race condition.

David-Baddeley, 2020-09-14:
hopefully fixed in 5f299dc53576a3ae6d4be7ccfb76bb0718ba6072

lukasfue, 2020-09-16:
This seems to still be an issue at least for splitterfitFNR for two-color ratiometric localizing. My estimate to localize a 50k frame dataset was about 12 hours, which is way longer than with the old launchworkers routine. I unplugged my network cable and that did seem to speed it up by about 30%, but still seems way longer than before. If memory serves me, I believe I used to be able to localize such a dataset in under an hour on py2 with launchworkers.

barentine, 2020-09-16:
Feel free to tap out at some point with debugging requests @lukasfue , but out of curiosity if you add `clusterIO-hybridns: False` to `.PYME/config.yaml` when your network cable is unplugged how fast is it?

Also worth checking how many fit tasks are actually being performed, rather than just number of frames.

Additionally it would be useful if once you stop the cluster you post pdf's of any of the mProf html outputs under `PYMEData/LOGS/`

sorry to be brief - thanks for trying this out!

lukasfue, 2020-09-16:
Since, my PC is currently off network I can check this out the next I'm in lab. 

How does one check number of fit tasks being performed?

I looked for those Logs when I aborted the last dataset but I couldn't find any.

David-Baddeley, 2020-09-16:
@lukasfue  - ouch! that sounds like something is going really, really wrong (even the 1hr for launch-workers sounds slow - I'd have expected something in the order of 10-20 mins on any reasonably modern hardware). It depends a lot on the localisation density and ROI size, but we were getting analysis frame-rates of 50-70 Frames/s back in 2013. 

Could you let me know what your ROI size is and how the data is formatted on disk (are you analysing from .h5, .tif, something else?). There is a chance that the ClusterOfOne stuff will be really bad if you are analysing directly from .tif or similar (this was already 2-3x slower with the old architecture, and has had extremely minimal testing with PYMEClusterOfOne).

lukasfue, 2020-09-16:
@David-Baddeley the density is quite high and the ROI is ~38x38 microns. The datasets being analyzed are all .dcimg files.

David-Baddeley, 2020-09-16:
What pixel size? You guys use ~100nm pixels right? Which would put it at ~ 380x380 pixels. Speed reference above was for roughly 200x450 pixels, so slightly smaller but not orders of magnitude out. The .dcimg might be a problem however, especially if it's a large .dcimg. I have a feeling the PYME dcimg loader pulls the whole file into memory even if you just want to access one frame. If you have a modern PC with, e.g. 12 cores, it'll be doing this simultaneously in 12 worker processes, which could exhaust your memory and force you into swap. To make things even worse, there's a chance that the files will be closed and re-opened for every frame. There are a whole bunch of file handling optimisations present for .h5/.pcs which are not there for other file formats.

It could be worth opening the .dcimg in dh5view and exporting as .h5 before doing the analysis, at least on one example. If this gives dramatically better performance, we could look at improving how .dcimg is handled.

lukasfue, 2020-09-16:
I'm not sure exactly off the top of my head, but yes around 100 nm. Okay, I will give that a try next I'm at my currently disconnected pc in lab.

Thanks!

lukasfue, 2020-09-18:
@David-Baddeley Okay, so I tried saving as a .h5 file and loading that to localize. I get the warning "ERROR: No metadata found in file ..." and "Carrying on with defaults - no guarantees it'll work well". Then another warning: "ADOffset fudged as 222 and probably wrong". Then if I try to localize, they all fail after the first 20 frames. I imagine this is because the metadata didn't get saved to the .h5 file for some reason?

@barentine As for `clusterIO-hybridns: False` with no network cable, it seemed to increase the time it would take by ~25% from what it takes with the network cable in and without that line.

barentine, 2020-09-25:
Hi @lukasfue let me know when you're working on this again and maybe it makes sense to do a remote desktop. Little alarming that the metadata wasn't getting saved to the h5. Once it's open in dh5view you can click `Modules > metadataView` to get a tab open if there wasn't one already and double check that the metadata is all there

lukasfue, 2020-10-02:
Hi @barentine, sorry this got put on the backburner for a bit there. So, the metadata definitely didn't save to the h5. All the values are wrong and its missing several values.

barentine, 2020-10-02:
No worries and thanks for circling back @lukasfue. I'm going to close this issue since we are now talking about a distinctly different one (kind of the third one in this thread). Would you mind opening a new issue about just the metadata not saving, and including the output of `type(image.data)` and then `type(image.mdh)` in the 'Shell' tab before you save?

closed 2020-10-02 by barentine