python-microscopy/python-microscopy #437: [BUG]

khu6, 2020-09-01:
**Describe the bug**
Localization fails for splitter fit

`Traceback (most recent call last):
  File "c:\users\kh725\python-microscopy\PYME\cluster\taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
Traceback (most recent call last):
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\cluster\taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
TypeError: 'dict_items' object is not subscriptable
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
TypeError: 'dict_items' object is not subscriptable
TypeError: 'dict_items' object is not subscriptable
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
TypeError: 'dict_items' object is not subscriptable
Traceback (most recent call last):
  File "c:\users\kh725\python-microscopy\PYME\cluster\taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
TypeError: 'dict_items' object is not subscriptable
Traceback (most recent call last):
  File "c:\users\kh725\python-microscopy\PYME\cluster\taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
TypeError: 'dict_items' object is not subscriptable
Traceback (most recent call last):
  File "c:\users\kh725\python-microscopy\PYME\cluster\taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "c:\users\kh725\python-microscopy\PYME\localization\remFitBuf.py", line 539, in __call__
    self.res[i] = fitFac.FromPoint(p.x, p.y)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\SplitterFitFNR.py", line 229, in FromPoint
    Xg, Yg, Xr, Yr, dataROI, bgROI, sigma, xslice, yslice, xslice2, yslice2 = self.getSplitROIAtPoint(x, y, z, roiHalfSize, axialHalfSize)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 196, in getSplitROIAtPoint
    DeltaX = self.shift_x.ev(x_, y_)
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 57, in shift_x
    self._shift_x = get_shiftmap(self.metadata['chroma.dx'])
  File "c:\users\kh725\python-microscopy\PYME\localization\FitFactories\FFBase.py", line 36, in get_shiftmap
    return twoColour.ShiftModel.from_md_entry(shiftmap)
  File "c:\users\kh725\python-microscopy\PYME\Analysis\points\twoColour.py", line 111, in from_md_entry
    cn, dict = json.loads(mdentry).items()[0]
TypeError: 'dict_items' object is not subscriptable`

**To Reproduce**
Steps to reproduce the behavior:
1. Start the cluster
2. Load splitter data
3. Add shiftmap
4. `Localize molecules`

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
Windows 10, Python 3.7.7, numpy 1.14.6
shiftmap generated using above

zacsimile, 2020-09-01:
Try changing line 111 to

```
cn, dict = list(json.loads(mdentry).items())[0]
```

We should probably also rename `dict` so the variable name isn't a Python type.

barentine, 2020-09-01:
Cool, thanks @khu6 for filing the issue and @zacsimile for the fix. @zacsimile you mind subbing the PR?

zacsimile, 2020-09-01:
Was waiting for confirmation that this does indeed fix it, but I'm fairly confident this does the trick so I'll submit it.

khu6, 2020-09-01:
Confirmation that Zach's fix solved this.

closed 2020-09-01 by khu6