python-microscopy/python-microscopy #442: [BUG] Trouble running PYMEAcquire analyze on Windows after spooling to pymeclusterofone

zacsimile, 2020-09-08:
**Describe the bug**
Spooled data to cluster. Hit STOP. Then pressed Analyze. dh5view started to come up.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch `pymeclusterofone`
2. In a separate window, launch `pymeacquire` (default init script)
3. Under Time/Blinking series, press "Start"
4. Wait for a few seconds, press "Stop"
5. Press "Analyze"

**Expected behavior**
`dh5view` should launch with my data

**Tracebacks**

<details>
<summary>pymeacquire</summary>
<pre>
Loading data
filename == PYME-CLUSTER:///scope_baby/2020_9_8/8_9_series_D.h5
Exception in thread zeroconf-ServiceBrowser__pyme-http._tcp.local._17380:
Traceback (most recent call last):
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\zeroconf\__init__.py", line 1684, in run
    state_change=service_type_state_change[1],
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\zeroconf\__init__.py", line 1438, in fire
    h(**kwargs)
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\zeroconf\__init__.py", line 1536, in on_change
    listener.add_service(*args)
  File "c:\users\scope_baby\code\python-microscopy\PYME\misc\pyme_zeroconf.py", line 115, in add_service
    if info.request(zc, 5000):
  File "c:\users\scope_baby\code\python-microscopy\PYME\misc\pyme_zeroconf.py", line 51, in request
    if None not in (self.server, self.address, self.text, self.port):
AttributeError: 'PatchedServiceInfo' object has no attribute 'address'

ERROR:PYME.IO.clusterIO:Timeout on get file http://127.0.0.1:53772/scope_baby/2020_9_8/8_9_series_D.h5/frame00000.pzf
Traceback (most recent call last):
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 421, in _make_request
    httplib_response = conn.getresponse()
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\http\client.py", line 1354, in getresponse
    response.begin()
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\http\client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\http\client.py", line 276, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\util\retry.py", line 403, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\urllib3\connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
[... 707 lines cut ...]
traits                    6.1.1            py36he774522_0
traitsui                  7.0.1                      py_0
ujson                     3.1.0            py36ha925a31_0
urllib3                   1.25.10                    py_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_3
wcwidth                   0.2.5                      py_0
webencodings              0.5.1                    py36_1
wheel                     0.35.1                     py_0
widgetsnbextension        3.5.1                    py36_0
win_inet_pton             1.1.0                    py36_0
wincertstore              0.2              py36h7fe50ca_0
winpty                    0.4.3                         4
wrapt                     1.12.1           py36he774522_1
wxpython                  4.0.4            py36ha925a31_0
xz                        5.2.5                h62dcd97_0
yaml                      0.1.7            vc14h4cb57cf_1  [vc14]  anaconda
zc.lockfile               2.0                        py_0
zeroconf                  0.17.0                   pypi_0    pypi
zeromq                    4.3.2                ha925a31_2
zipp                      3.1.0                      py_0
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
zstd                      1.4.5                h04227a9_0
</pre>
</details>

barentine, 2020-09-08:
@zacsimile can you try updating zeroconf to 0.17.7? Can't remember off the top of my head which version the `PatchedServiceInfo` init gets changed (and I should probably do a little legwork to PR an update to the last zeroconf version with py27 support, 0.19.1. Doing so would let py3 users use 0.21.3 (or later maybe) which could help with some packaging (someone sent me an email with conda not pulling zeroconf 0.17.7 for py37)

David-Baddeley, 2020-09-14:
@zacsimile - hmm, that error is odd - 0.17.0 (as in `conda env list`) should work fine, I think, I wonder if conda / pip is lying to you and you actually have a more recent version installed. 

There's a chance that the zeroconf issues are actually a red herring here though, as a) cluster of one uses sqlite based resolution by default and b) the error happens in the zeroconf service browser thread (meaning that everything else should continue fine). Have you also looked in the `LOGS` directory (tends to have a bit more info than lands on the console for the cluster based analysis)?

Medium term zeroconf strategy as follows:
- Keep Py27 pinned against 0.17[.7] 
- Bump to latest release on py3
- submit a patch against zeroconf to allow running on a non-standard port (see https://github.com/jstasiak/python-zeroconf/issues/186) [this has been sitting stalled as I don't see any point patching against an old version]. In general zeroconf / PYMEClusterOfOne on Win10 on a moderately busy network == lots of problems.

Quick zeroconf / cluster of one Win10 semi-workaround is to unplug the network cable and see if that resolves things.

barentine, 2020-10-13:
I created a new conda environment today to see if some cam-ready errors in tiling might be due to non-mkl linked numpy slowdown, and ran into the same `AttributeError: 'PatchedServiceInfo' object has no attribute 'address'`

<details> <summary>conda list</summary>

```
# packages in environment at C:\Users\Bergamot\Anaconda2\envs\chaos:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py36_0
babel                     2.8.0                      py_0
blas                      1.0                         mkl
blosc                     1.20.0               h7bd577a_0
brotlipy                  0.7.0           py36he774522_1000
bzip2                     1.0.8                he774522_0
ca-certificates           2020.7.22                     0
cairo                     1.14.12              hf171d8a_3
certifi                   2020.6.20                py36_0
cffi                      1.14.3           py36h7a1dbc1_0
chardet                   3.0.4                 py36_1003
cheroot                   8.4.5                    py36_0
cherrypy                  18.6.0                   py36_0
cloudpickle               1.6.0                      py_0
cryptography              3.1.1            py36h7a1dbc1_0
cycler                    0.10.0           py36h009560c_0
cython                    0.29.21          py36ha925a31_0
cytoolz                   0.11.0           py36he774522_0
dask-core                 2.30.0                     py_0
decorator                 4.4.2                      py_0
dispatch                  1.0                      py36_0    david_baddeley
docutils                  0.16                     py36_1
fftw                      3.3.4                         0    david_baddeley
freetype                  2.10.3               hd328e21_0
future                    0.18.2                   py36_1
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                 ha925a31_3
idna                      2.10                       py_0
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0                      py_0
imagesize                 1.2.0                      py_0
importlib-metadata        2.0.0                      py_1
importlib_metadata        2.0.0                         1
importlib_resources       1.4.0                    py36_0
intel-openmp              2020.2                      254
jaraco.classes            3.1.0                      py_0
jaraco.collections        3.0.0                      py_0
jaraco.functools          3.0.1                      py_0
jaraco.text               3.2.0                    py36_1
jinja2                    2.11.2                     py_0
joblib                    0.17.0                     py_0
jpeg                      9b                   hb83a4c4_2
kiwisolver                1.2.0            py36h74a9793_0
libpng                    1.6.37               h2a8f88b_0
libtiff                   4.1.0                h56a325e_1
lz4-c                     1.9.2                hf4a77e7_3
lzo                       2.10                 he774522_2
markupsafe                1.1.1            py36he774522_0
matplotlib                3.3.1                         0
matplotlib-base           3.3.1            py36hba9282a_0
mkl                       2020.2                      256
mkl-service               2.3.0            py36hb782905_0
mkl_fft                   1.2.0            py36h45dec08_0
mkl_random                1.1.1            py36h47e9c7a_0
mock                      4.0.2                      py_0
more-itertools            8.5.0                      py_0
mpld3                     v0.3                     py36_0    david_baddeley
networkx                  2.5                        py_0
numexpr                   2.7.1            py36h25d0782_0
numpy                     1.19.1           py36h5510c5b_0
numpy-base                1.19.1           py36ha3acd2a_0
[... 65 lines cut ...]
wxpython                  4.0.4            py36ha925a31_0
xz                        5.2.5                h62dcd97_0
yaml                      0.2.5                he774522_0
zc.lockfile               2.0                        py_0
zeroconf                  0.28.0                   py36_3    david_baddeley
zipp                      3.3.0                      py_0
zlib                      1.2.11               h62dcd97_4
zstd                      1.4.5                h04227a9_0

```

</details>

Was getting ready to carry out the plan David outlined above, and in doing the version check on zeroconf noticed
```
In [3]: zeroconf.__version__
Out[3]: '0.24.0'
```
in the same conda environment.  Taking a peak at their git, the tag for 0.28.0 definitely looks like it would pull/tag itself right
[image]

I'll try pip'ing the latest zeroconf and report back. 

other notes:
turns out I didn't have clusterIO-hybridns config'd to False on my instrument computer (I think I'd taken it out for some testing).

barentine, 2020-10-13:
ah, note the 0.24 / 0.28 thing was probably just due to not having ipython kernel installed in that new conda environment and pulling an old one

barentine, 2020-10-13:
Note that current head seems to run fine with zeroconf 0.26.3 (which makes sense given the address param for service info isn't dropped until 0.27.0)

barentine, 2020-11-18:
OK to close @zacsimile or any updates here? I agree with David this must have been a zeroconf issue.

zacsimile, 2020-11-18:
Yes, I've gotten `pymeclusterofone` to work since this. Will open a new issue if this comes back.

closed 2020-11-18 by zacsimile