python-microscopy/python-microscopy #455: [ENHANCEMENT] numbers on LUT

barentine, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I want to snapshot someone an image and i have to include the layer menu histogram in order to communicate what the units/min-max of the LUT are

[image]

**Describe the solution you'd like**
numbers top/bottom like people are used to. Maybe a checkbox up near the LUT / scale bar to toggle including the numbers / scale bar unit in the canvas?

**Are you prepared to work on it?**
sure thing

David-Baddeley, 2020-09-25:
Sounds reasonable.

Implementation might not be very fun (text in Opengl is done by displaying
images of the individual characters with very little high level support).
Had something working for text in the pre-shaders code, but it gets even
more complicated when using shaders.

Probably best to implement generic text display logic first, then use that
from the LUT. Will need a bit of work to get scaling etc sensible with
translated coordinate systems.

David-Baddeley, 2020-09-29:
I wonder if the right approach might not be to generate images of the complete text, using, e.g. PIL. It'll be next to impossible to get kerning right if we go with the single character approach ... (unless we use a monospace font, but that is aesthetically unappealing).

zacsimile, 2021-05-18:
Text rendering described at https://learnopengl.com/In-Practice/Text-Rendering doesn't seem too bad. They seemed to have solved the kerning problem.

David-Baddeley, 2021-05-18:
Freetype looks like it has the potential to be a pain to get installed on (and packaged for) windows. We might be able to render an entire text string to a bitmap using either PIL or cairo, and display that as a texture on a single quad without having to worry about dependencies etc (or kerning).

David-Baddeley, 2021-05-18:
We might be able to use wx to draw the bitmap - that way we can easily match the font to the font used for other display items. 

```python
 @classmethod
  def gen_text_image(cls, text='', size=12, font='courier'):
      # TODO - implement size, font, etc ...
      import wx
      import numpy as np
      
      dc = wx.MemoryDC()
      w, h = dc.GetTextExtent(text)
      bmp = wx.Bitmap(w, h)
      dc.SelectObject(bmp)
      dc.SetTextForeground(wx.WHITE)
      dc.DrawText(text, 0, 0)
      dc.SelectObject(wx.NullBitmap)
      
      im = np.zeros([h, w, 3], dtype='uint8')
      bmp.CopyToBuffer(im.data)
      
      return im
```

The image of the complete text can then be bound to a texture and mapped to a quad, rather than having to do it character by character.

David-Baddeley, 2021-05-18:
altogether something like (still having a lot of fixmes, as well as missing a shader etc):

```python
class Text(object):
    def __init__(self, text='', pos=(0,0)):
        import Image
        self._texture_id = None
        self._img = None

        self._pos = pos
        #just take red channel
        self._im = np.ascontiguousarray(self.gen_text_image(text)[:,:,0])
        self._h, self._w = self._im.shape
        
        
    @classmethod
    def gen_text_image(cls, text='', size=12, font='courier'):
        # TODO - implement size, font, etc ...
        import wx
        
        dc = wx.MemoryDC()
        w, h = dc.GetTextExtent(text)
        bmp = wx.Bitmap(w, h)
        dc.SelectObject(bmp)
        dc.SetTextForeground(wx.WHITE)
        dc.DrawText(text, 0, 0)
        dc.SelectObject(wx.NullBitmap)
        
        im = np.zeros([h, w, 3], dtype='uint8')
        bmp.CopyToBuffer(im.data)
        
        return im
        
        
    
    def set_texture(self, image):
        if self._texture_id is None:
            self._texture_id = glGenTextures(1)
        
        if image is None:
            return
        
        if not image is self._img:
            self._img = image
            
            #image = image.T.reshape(*image.shape) #get our byte order right
            
            glActiveTexture(GL_TEXTURE0)
            glBindTexture(GL_TEXTURE_2D, self._texture_id)
            glPixelStorei(GL_UNPACK_ALIGNMENT, 1)
            glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_S, GL_CLAMP)
            glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_T, GL_CLAMP)
            glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MAG_FILTER, GL_NEAREST)
            #glTexParameteri (GL_TEXTURE_2D, GL_TEXTURE_MAG_FILTER, GL_LINEAR)
            glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MIN_FILTER, GL_NEAREST)
            #glTexParameteri (GL_TEXTURE_2D, GL_TEXTURE_MIN_FILTER, GL_LINEAR)
            glTexEnvf(GL_TEXTURE_ENV, GL_TEXTURE_ENV_MODE, GL_REPLACE)
            glTexImage2D(GL_TEXTURE_2D, 0, GL_R32F, image.shape[1], image.shape[0], 0, GL_RED, GL_FLOAT,
                         image.astype('f4'))
    
    def render(self, gl_canvas):
        with self.shader_program:
            self.set_texture(self._im)
        
            glEnable(GL_TEXTURE_2D) # enable texture mapping */
            glActiveTexture(GL_TEXTURE0)
            glBindTexture(GL_TEXTURE_2D, self._texture_id) # bind to our texture, has id of 1 */
            glUniform1i(self.shader_program.get_uniform_location("im_sampler"), 0)
        
            # FIXME - choose appropriately for current viewport - want to make it so that the text renders real size
            # (i.e. unwind any model-view / and projection stuff).
            x0, y0 = self._pos
            x1 = x0 + self._w #*scale
            y1 = y0 + self._h #*scale
        
            glDisable(GL_TEXTURE_GEN_S)
            glDisable(GL_TEXTURE_GEN_T)
            glDisable(GL_TEXTURE_GEN_R)
        
            #glColor3f(1.,0.,0.)
            glColor4f(1., 1., 1., 1.)
            glBegin(GL_QUADS)
            glTexCoord2f(0., 0.) # lower left corner of image */
            glVertex3f(x0, y0, 0.0)
            glTexCoord2f(1., 0.) # lower right corner of image */
            glVertex3f(x1, y0, 0.0)
            glTexCoord2f(1.0, 1.0) # upper right corner of image */
            glVertex3f(x1, y1, 0.0)
            glTexCoord2f(0.0, 1.0) # upper left corner of image */
            glVertex3f(x0, y1, 0.0)
            glEnd()
        
            glDisable(GL_TEXTURE_2D)
```

David-Baddeley, 2021-05-19:
Prototype functionality now in #1038.

[image]

Not really sure I like it though - there's not really enough space for the labels, and it adds clutter. Font rendering could also be improved (maybe using alpha in the shader was the wrong call, or maybe it's just an OSX pixel scaling thing). We'd have to a) make it switchable and b) make it robust for different layer types (which didn't have/use clim) before merging.

David-Baddeley, 2021-05-19:
now disabled by default in #1038, enable by entering the following in the shell

```python
>>> glCanvas.LUTOverlayLayer.show_bounds = True
```

David-Baddeley, 2021-05-19:
Gets really messy when you have multiple layers.

zacsimile, 2021-05-19:
Looking good, @David-Baddeley! Looking at the example you showed, can we try right-justifying the text? Hopefully that will get the full number on screen. I like having it off by default. It may be useful to include a GUI toggle, even if it's just under "Extras".

David-Baddeley, 2021-05-19:
Right-justification becomes super-problematic when you have multiple layers (puts the labels above the wrong bar), but it might be worth a shot - it's pretty much unusable with multiple layers in any case.

zacsimile, 2021-05-19:
To do multiple layers we'd probably need to shrink the height of the LUTs and run the numbers at 90 degrees above and below the LUT so everything is in a single column.

David-Baddeley, 2021-05-19:
or just draw the  vertically oriented text over the top of the LUT ...

David-Baddeley, 2021-05-19:
Will merge #1038, can then refine later.

zacsimile, 2021-05-20:
Nice! Works as expected on my machine. Definitely need to play with the style/orientation of the bounds but overall good progress.

I worry drawing the vertically oriented text over the top of the LUT could be rather hard to see depending on the LUT.