python-microscopy/python-microscopy #460: [BUG] PYMEVis will color layers on `z` by default even if it is not present

zacsimile, 2020-09-29:
**Describe the bug**
If we open a MATLAB file and only specify columns x and y, the data will load, but the display will appear black because the default `Colour` option for `Layer 0 - output` will be set to `z`.

**Expected behavior**
The points should display, perhaps we need to color on another variable by default in the absence of `z` and `t`.

**Screenshots**
[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.6
 - Python version: 3.6.10
 - numpy version: 1.16.0
 - wxpython version [if a UI bug]: 4.0.4

yongdengzhang, 2020-09-29:
When I tried to open a 4Pi data for the first time, I didn't see z in the drop-down menu so it was not specified. Then after loading, no image was displayed because z was the default color option. Even if this bug is fixed, I will still forget to specify z as I have to manually type z in the box, which is not something straightforward for a beginner. It would be great to add z in the drop-down menu but not set it as a required variable.

David-Baddeley, 2020-09-29:
Hmm, this smells a little like a regression - If both z and t were absent I'm pretty sure we used to set the colour to `''` (or possibly `'<none>'`and it would display as a constant colour.

zacsimile, 2020-09-29:
I also thought we'd fixed this a while ago.

zacsimile, 2020-09-29:
I think the short answer is the `pipeline` is returning  a `z` key even when I don't define one.

David-Baddeley, 2020-09-29:
that should not be a problem - the pipeline should always have a `z` key and it should default to 0 if `z` is not in the data. Maybe it's the defaulting to 0 which is the problem (I suspect this currently relies on `t` being present, but would need to double check).

zacsimile, 2020-09-30:
You are right that `z` gets generated in the pipeline with values of 0. I forgot this was default behavior. There's no `t` in my pipeline. Here's what I think happens:

- In `PYME.LMVis.visCore._create_base_layer`, it checks for `t` and then `z`. Finding `z`, it sets this as `vertexColour`.
- In `PYME.LMVis.layers.pointcloud.update_from_datasource`, it sets `c=ds['z']`, which is all zeros.

It looks like if there was no color key present, it would also default all color values to 0 (line 192 of `PYME.LMVis.layers.pointcloud`). I have three potential solutions:

1. In `update_from_datasource`, check if `c` is all 0s. If so, set `c` to all 1s and maybe set alpha to 0.5 so it's clear something's up?
2. In `_create_base_layer`, if `z` is all 0s, set `vertexColour=''` and change the behavior of `vertexColour=''` in `update_from_datasource` to something like solution 1
3. In `_create_base_layer`, if `z` is all 0s, sidestep the problem by setting `cmap='C'`

What do you think?

David-Baddeley, 2020-09-30:
I wonder if this is a colour scaling issue rather than anything else - the default colour map should still generate a solid colour if it's all zeros. The histLimits editor may not be as cooperative.

David-Baddeley, 2020-10-09:
Hmm, cant't reproduce with .csv. Also not a py3 issue. See attached file with only x and y columns. Is it possible that there is something odd in the matlab file (e.g. a z variable of the wrong size).

[test_xy.csv.zip](https://github.com/python-microscopy/python-microscopy/files/5351977/test_xy.csv.zip)

[image]

zacsimile, 2020-10-09:
I can reproduce with a CSV.

[image]

I think it comes down to all 0s displaying as black.

[image]

David-Baddeley, 2020-10-09:
That is super, super odd as I don't get that - you could get it with a few of the LUTs (e.g. spectral) but shouldn't with a rainbow LUT (0s should be red). Wonder if it's a matplotlib version issue ... but even then it would still seem odd. I explicitly tried both py3 and py3 in case there was something going on there.

David-Baddeley, 2020-10-09:
* get points showing up red in both cases. Matplotlib 3.0.2 on py3 (which isn't super old).

zacsimile, 2020-10-09:
All new installs come with `matplotlib>=3.2.2`. I'm running `3.3.0`. Downgraded to `3.0.2` and it now shows up red. I think this is a problem we can expect to run into in the future, though. Maybe we just change the default color value to 255 instead of 0?

David-Baddeley, 2020-10-19:
I don't think changing the default z value is really a solution, having z=0 for a 2D image is intuitive and what you *should* get.  The point is that these points are not displaying with a default value. They are displaying z, z has a sensible value, and that value should map to a visible point. We need to work out what changed in recent matplotlib installs and find a fix / workaround for that.

David-Baddeley, 2020-10-19:
I wonder if it's a floating point precision issue meaning that we map to a very slightly negative value and matplotlib has got strict with it's colour bounds).

zacsimile, 2020-11-05:
How would we test if it's a floating point issue? We ascribe a value of `0`, so I wonder where the negative value comes from.

zacsimile, 2020-11-18:
This is not an error for me anymore. I'm not sure what changed. I'm running `matplotlib==3.3.0` and it works just fine now.

David-Baddeley, 2020-11-27:
Added needs verification tag as it's a) unclear if it's still an issue and b) unclear why it was ever an issue - i.e. we haven't explicitly done anything to fix

zacsimile, 2020-12-04:
Just kidding. This is still an issue. However, once I click anywhere in the layer histogram, the points appear in the glCanvas. If I don't click in the layer histogram, and I change the background color to white, the spots appear black on a white background as before.

zacsimile, 2020-12-07:
Fixed in #670.

closed 2020-12-07 by zacsimile