python-microscopy/python-microscopy #461: [BUG] Shortcut 'C' does not center the current ROI image after View Clipping

yongdengzhang, 2020-09-29:
**Describe the bug**
If I use View Clipping to select an ROI region (for multicolor data), the shortcut 'C' will not center the current image.

**Expected behavior**
The range defined by View Clipping should be applied to all color channels, and press 'C' should center the current ROI.

David-Baddeley, 2020-09-29:
This  behaviour is expected, if not entirely intuitive and is a result of the view clipping and ROI functions being separate but having some overlap in what they can achieve. The `c` shortcut centres on the current ROI. It is probably best addressed by improving documentation, although adding some form of additional 'centre on clipped region' function might be possible. The difference between the ROI and view clipping functions is outlined briefly below:

ROI
---

- The default way of restricting the field of view
- Restricts both what is displayed and what is used for subsequent processing (generating images, clustering, etc ...)
- Operates on the CPU, and is not super-fast (not really fast enough for real-time slicing on large datasets)
- Currently only restricts x & y

View Clipping
--------------

- Restricts display only (any analysis uses all points in the ROI) 
- Implemented to permit fast slicing and animation - e.g. displaying a thin xy  or xz slice and scrolling through
- Happens at the last possible moment (on GPU, after data is uploaded)

There's an argument that most of the times that you'd want to centre on the clipped region (i.e. you are not dynamically changing it) it would be more appropriate to use a ROI to restrict the region rather than the clipping function.

The current view clipping vs ROI stuff is an unfortunate consequence of aiming for high performance and is not super intuitive. We should have a think about how to improve this.

David-Baddeley, 2020-09-29:
Changed to enhancement as it's not technically a bug (not that we shouldn't find a way of addressing it)

zacsimile, 2020-09-29:
From personal observation, one of the strange things is if I have multiple layers, and I view clip on composite layer, `c` centers the region. If I then flip to a different layer, the view clipping settings remain the same, but the entire layer is visible and unclipped. If I then flip back to the composite layer as visible, `c` no longer works. If I now reset view clipping to full range in both the composite and the other layer, and then re-clip the composite layer, `c` works again.

David-Baddeley, 2020-09-29:
What do you mean by a composite layer? The clipping *should* effect all layers equally, although as it's implemented in the vertex shader it is possible that some types of layers don't respect it (surfaces and images come to mind, although I think I've fixed one of those).

zacsimile, 2020-09-29:
Yeah, but it doesn't effect all layers equally right now, is what I'm saying. At least not on my machine (macOS 10.15 py3.6 latest master code). I think this might be a bug.

David-Baddeley, 2020-09-29:
@zacsimile - what you are running into is a separate issue, and relates to having layers which have a different extent (bounding box). As it stands, we reset the view clipping to include all data whenever the bounding box of the displayed layers changes (e.g. by toggling layer visibility when you have different sized layers). 

This behaviour is justified with the rationale that:
- if you, e.g. add or enable a layer, you generally want to see it.
- if the bounding box is not the same as the layers already displayed then whatever clipping has been applied is likely
not sensible, so we should give up and just go to a safe default clipping which shows everything. 
- if you change the ROI you generally want that to reflect in what you see (ie changes to the ROI should have precedence over whatever you've set the clipping bounds to be)
- When I wrote it I probably assumed that changes in bounding box size will most likely be the result of changes in ROI (although this is probably not strictly correct)

The behaviour is potentially a bit too conservative, although it's a hard one to get right.

zacsimile, 2020-09-29:
Got it. I think that should be fine. In practice, I think the bounding box gets reset, but I don't think that change propagates properly to view clipping, which might be where some of the confusion comes in from a user perspective. I know this is why I thought these errors were linked. For example, I can view clip one layer, and then open another layer, and now the bounding box is reset, but the view clipping isn't. 

[image]

David-Baddeley, 2020-09-29:
OK - it looks like there is a legitimate update issue where the view clipping panel doesn't get updated if you change the view clipping elsewhere in the code. This is somewhat tangential to the behaviour of the `C` key.

zacsimile, 2020-11-20:
This separate view-clipping issue is fixed in commit dc54a96a04fc024feedcd1601cc574385de3f36a of PR #586.

David-Baddeley, 2020-11-21:
@zacsimile - dc54a96 hasn't fixed the lack of updates per-se, it's just removed the most likely place where it might manifest.