python-microscopy/python-microscopy #463: [ENHANCEMENT]

David-Baddeley, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
Pixel size [nm] box in importers is confusing (see https://publish.twitter.com/?query=https%3A%2F%2Ftwitter.com%2Fchristlet%2Fstatus%2F1311415653681836032&widget=Tweet)

This has been discussed previously but I don't believe it's been specifically flagged as an issue

**Describe the solution you'd like**
Two possible solutions:

1)
- just rename the parameter

2)
- Have 2 radio buttons, "Localisations are in units of nm" (default) and "Localisations are in units of pixels"
- enable the pixel size box if the user selects "Localisations are in units of pixels"

barentine, 2020-09-30:
@David-Baddeley you were reviewing a PR with this fix today. #284 changes the display name and adds helptext, leaving the underlying parameter the same so old recipes still work.

barentine, 2020-09-30:
from that PR, and similarly in the twitter thread:
[image]

zacsimile, 2020-09-30:
Is it `nanometer units` or `nanometers-per-unit`?

David-Baddeley, 2020-09-30:
PR mentioned addresses use in recipes but not (yet) the text import dialog. I remember us having a really hard time finding a short name which sufficiently conveyed what it did (I think 'nanometer units', although an improvement is not also not completely intuitive).  Suggest we run with the current / new naming in the recipe module, but still consider the 2 step approach for the import dialog (as that's really intuitive / hard to get wrong).

barentine, 2020-10-01:
Great point David - sorry i wasn't slowing down to read carefully. The two-step radio-button on import sounds like a great way to make sure it is clear.

barentine, 2020-10-09:
With #284 and #496 both merged I think we can close this

closed 2020-10-18 by David-Baddeley