python-microscopy/python-microscopy #468: [BUG] metadata not saving to h5 files in dh5view

lukasfue, 2020-10-02:
**Describe the bug**
I loaded some raw data as a DCIMG file with metadata in a JSON file into dh5view. I saved that as an H5 file and then tried to load that H5 file into dh5view, but get an error that the metadata did not load properly and indeed it doesn't when I check the metadata tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Load DCIMG raw data in dh5view
2. Save localizations as H5 file
3. Load H5 into new dh5view 
4. See error about metadata
5. Comparing original metadata in JSON file to Metadata tab shows they are different

**Expected behavior**
No issues with loading metadata

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.8.3
 - numpy version: 1.19.1

**Additional context**
type(image.data) output: <class 'PYME.IO.DataSources.BGSDataSource.DataSource'>
type(image.mdh) output: <class 'PYME.IO.MetaDataHandler.NestedClassMDHandler'>

barentine, 2020-10-02:
Thanks for posting @lukasfue! Just to double check - there's nothing printed in the console when you save the series to h5?

lukasfue, 2020-10-02:
Disregard outputs on initial comment. These are the correct ones:

type(image.data) output: <class 'PYME.IO.DataSources.DcimgDataSource.DataSource'>
type(image.mdh) output: <class 'PYME.IO.MetaDataHandler.NestedClassMDHandler'>
image.mdh output: 
0.005009
0.005009
0
<NestedClassMDHandler>:

voxelsize.units: um
voxelsize.x: 0.099
voxelsize.y: 0.099
voxelsize.z: 0
Camera.ReadNoise: 1
Camera.NoiseFactor: 1
Camera.ElectronsPerCount: 0.47
Camera.TrueEMGain: 1
Camera.ADOffset: 100
Camera.CycleTime: 0.005009
Camera.IntegrationTime: 0.005009
Camera.ROIOriginX: 0
Camera.ROIOriginY: 0
Camera.ROIHeight: 400
Camera.ROIWidth: 880
Camera.ROIX0: 0
Camera.ROIY0: 0
Camera.SensorHeight: 2048
Camera.SensorWidth: 2048
Camera.SerialNumber: 100233
tIndex: 0
SeriesPattern: AF647-100_CF660C-100_Cell1*.dcimg
StackSettings.FramesPerStep: 50000
StackSettings.NumCycles: 1
StackSettings.NumSteps: 1
StackSettings.StepSize: 0.0
StackSettings.TossFramesBetweenSteps: False
Sample.Labelling: []
Splitter.Channel0ROI: [0, 0, 400, 400]
Splitter.Channel1ROI: [480, 0, 400, 400]
Splitter.Dichroic: FF01-709/167
Splitter.Flip: False
Splitter.TransmittedPathPosition: Right
EstimatedLaserOnFrameNo: 0
imageID: 0
StartTime: 0
voxelsiz.units: um

command prompt output after I save:
WARNING:PYME.DSView.modules:Plugin [arrayView] injects into dsviewer namespace, could result in circular references
WARNING:PYME.DSView.modules:Plugin [shell] injects into dsviewer namespace, could result in circular references
0
WARNING:PYME.DSView.modules:Plugin [metadataView] injects into dsviewer namespace, could result in circular references
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.processing
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.processing
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.io
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.io
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.chung_cc
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.chung_cc
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.localisations
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.localisations
0
WARNING:PYME.DSView.modules:Plugin [recipes] injects into dsviewer namespace, could result in circular references
Creating fold panel
c:\users\laf62\python-microscopy\PYME\DSView\scrolledImagePanel.py:52: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\laf62\python-microscopy\PYME\DSView\arrayViewPanel.py:1154: wxPyDeprecationWarning: Call to deprecated item ImageFromData. Use :class:`Image` instead.
  img = wx.ImageFromData(ima.shape[1], ima.shape[0], ima.ravel())
c:\users\laf62\python-microscopy\PYME\DSView\arrayViewPanel.py:513: wxPyDeprecationWarning: Call to deprecated item BitmapFromImage. Use :class:`wx.Bitmap` instead
  im2 = wx.BitmapFromImage(im)
c:\users\laf62\python-microscopy\PYME\DSView\scrolledImagePanel.py:52: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\laf62\python-microscopy\PYME\DSView\arrayViewPanel.py:513: wxPyDeprecationWarning: Call to deprecated item BitmapFromImage. Use :class:`wx.Bitmap` instead
  im2 = wx.BitmapFromImage(im)
(880, 400)

barentine, 2020-10-02:
Thanks for rerunning before saving. 

Quick double check - when you say 'save localizations to h5' do you mean localizations, or the image series? I'm assuming you are trying to save the image series to h5 for testing the speed of fitting when its saved as h5 vs saved as dcimg. It's a bit weird to guess where this is coming from since I'm not having any problems saving to h5 with metadata using either `File > Save` or `File > Save as`, but I haven't tested with a dcimg yet.  I'm guessing the dcimg isn't particularly small, so maybe send me a google drive link or so if it isn't too much trouble. Otherwise I'll suggest some more things you can print out and we can keep iterating

lukasfue, 2020-10-02:
Yeah, sorry I was thinking of Visgui in my head. I did the File > Save > h5 file. Okay, do you want the exact dcimg i had the issue with? Its on the larger side.

barentine, 2020-10-02:
If it's easy enough for you to upload it I don't mind, any dcimg as long as you know you have the issue with it.

It's a bit bizarre, since I am also saving from a `NestedClassMDHandler`. Would you mind running the following? (or just send me the dcimg and I'll play with it).
all in the shell, one line at a time, where r'path-to-test-file.h5' is wherever it's convenient for you to put something, e.g. r'C:\lukas\Desktop\test.h5'
```
import tables
from PYME.IO.MetaDataHandler import HDFMDHandler

h5out = tables.open_file(r'path-to-test-file.h5', 'w', chunk_cache_size=2**23)
outMDH = MetaDataHandler.HDFMDHandler(h5out)
outMDH.copyEntriesFrom(image.mdh)
h5out.flush()
h5out.close()
```
and then open that file and in shell `image.mdh`

David-Baddeley, 2020-10-02:
Where did you set the metadata? From within dh5view, or in a metadata.json
file? The metadata editing within dh5view happens on copy, so this could be
the problem.

barentine, 2020-10-05:
@David-Baddeley I'm guessing it gets set for him automatically loading the dcimg into dh5view by `image.ImageStack._findAndParseMetadata` which looks to just load the json and `self.mdh.update(<the-loaded-dict>)`. I would think everything being accessible to the `NestedClassMDHandler` would mean making an `HDFMetaDataHandler` and calling the `copyEntriesFrom` method should be fine.

@lukasfue any updates here? Did you upload some test data somewhere? I'm bugfixing the rest of the afternoon if you want me to take a crack at anything. 

Might also be worth having a look at the output of conda list. Mind doing something like `conda activate pyme` followed by `conda list > lukasfue-conda-20201004.txt`? and uploading that here? You're on a later version of python and numpy than I've been using so far, which might mean you've got different versions of the package in charge of opening/saving hdf5 files. Maybe also upload your saved image h5 if you can? Or when you open it in dh5view, go to shell and type `image.dataSource.h5File`
You should get something like
 <details><summary> this </summary> 

```
File(filename=/media/smeagol/ae1130e6-c525-47fe-adec-f6af8e76c0c7/2020_9_28_psf1_avg_resampled.h5, title='', mode='r', root_uep='/', filters=Filters(complevel=0, shuffle=False, bitshuffle=False, fletcher32=False, least_significant_digit=None))
/ (RootGroup) ''
/Events (Table(0,), shuffle, zlib(5)) ''
  description := {
  "EventDescr": StringCol(itemsize=256, shape=(), dflt=b'', pos=0),
  "EventName": StringCol(itemsize=32, shape=(), dflt=b'', pos=1),
  "Time": Time64Col(shape=(), dflt=0.0, pos=2)}
  byteorder := 'little'
  chunkshape := (221,)
/ImageData (EArray(230, 1024, 256), shuffle, zlib(1)) ''
  atom := Float64Atom(shape=(), dflt=0.0)
  maindim := 0
  flavor := 'numpy'
  byteorder := 'little'
  chunkshape := (1, 1024, 256)
/MetaData (Group) ''
/MetaData/Camera (Group) ''
/MetaData/CameraOrientation (Group) ''
/MetaData/Lasers (Group) ''
/MetaData/Multiview (Group) ''
/MetaData/Positioning (Group) ''
/MetaData/Protocol (Group) ''
/MetaData/StackSettings (Group) ''
/MetaData/cropping (Group) ''
/MetaData/voxelsize (Group) ''
/MetaData/Lasers/MPB560 (Group) ''
/MetaData/Lasers/MPB642 (Group) ''
/MetaData/Lasers/OBIS405 (Group) ''
/MetaData/Lasers/OBIS488 (Group) ''

```
</details>

lukasfue, 2020-10-06:
@David-Baddeley @barentine  The metadata was set in a metadata.json file

So, I was going to upload some data for you to test this with, but I'm observing some inconsistency. The file it was occurring with before, still has the same metadata issue. However, if I save a new h5 file from the original dcimg/json (file/metadata), I do not get an error anymore. Hopefully, that made sense.

barentine, 2020-10-08:
Cool, if I understand correctly it sounds like one time there was a problem saving an open series to h5 (it got saved without metadata and only has some defaults on load), and you haven't run into it on subsequent saves? If so, @lukasfue feel free to close this for now and re-open it if it happens again.

barentine, 2020-10-13:
@David-Baddeley or @lukasfue mind closing this?

closed 2020-10-19 by David-Baddeley