python-microscopy/python-microscopy #471: [BUG] Can't open two files in a row with different pipeline variables on py3

zacsimile, 2020-10-04:
**Describe the bug**
PYMEVis won't let me open one file, and then let me decide to use and open different file, without first launching a new instance, if the second file does not contain the same pipeline variables as the first.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any file in PYMEVis.
2. Try to open a different file that has fewer pipeline variables than the first file.

**Expected behavior**
The second file replaces the first file in my PYMEVis window.

**Traceback**
```
 Error whilst running File>&Open
==============================================
python-microscopy=20.09.18
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 736, in OpenFile
    self.pipeline.OpenFile(filename, **args)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 828, in OpenFile
    self.recipe.execute()
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/tablefilters.py", line 38, in execute
    filtered = tabular.ResultsFilter(inp, **self.filters)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 752, in __init__
    raise KeyError('Requested key not present: ' + k)
KeyError: 'Requested key not present: error_y'
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
Add any other context about the problem here.

barentine, 2020-10-04:
Call me crazy, but this might be solved with merge of #284.

David-Baddeley, 2020-10-05:
This is due to the fact that opening a new file doesn't clear the current recipe. The old `FilterTable` module remains in the pipeline (with it's old keys). This should only manifest if the missing keys are ones which would normally be filtered on (e.g. `error_y` as with the traceback).

A quick short-term fix that will still work with #284 might be to clear/delete the recipe (`pipeline.recipe.modules = [];pipeline.clear()`) at the start of `OpenFile()`.  This will however clobber any recipe that is already loaded or we have created with the `-r <recipe_name>` command line option. This is arguably already a bit broken (status quo is to add extraneous modules to the existing recipe when we call `OpenFile()`. As it stands, loading a recipe on the command line only works if the filename is specified on the command line (the recipe is loaded after the file and clobbers anything that `OpenFile()` adds. Clobbering a loaded recipe might be the most intuitive solution (`File-->Open` being pretty much a hard reset).

Alternative solutions would be some combination of the following: 

- Check to see if there is already a recipe defined and not add modules for `ProcessColour` and `FilterTable` if that is the case.
- Check, and if needed drop any undefined filter keys (yuck)
- Make the filter more tolerant to missing keys (e.g. printing an error message, but ignoring any missing keys)

David-Baddeley, 2020-10-07:
I think we can safely close now ...

closed 2020-10-07 by David-Baddeley