python-microscopy/python-microscopy #473: [ENHANCEMENT] Non-filename templated recipe parameters

barentine, 2020-10-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I have a templated recipe that needs a non-file str parameter changed between samples. While this is usually somewhere in the directory and could be grabbed automatically from the URI, that would take a special casing to ignore e.g. `analysis` directory, chunked `000`-type directories, etc. 

[image]

**Describe the solution you'd like**
While a web-UI for full editing would be neat, a stop-gap would be if we special-cased another e.g. `USERSTR` parameter name which was exposed in the same menu but didn't prepend `pyme-cluster:///`.

**Additional context**
Running this automatically from the scope w/chaining isn't a big deal (but requires tweaking the recipe for each sample) but I do find it a bit annoying to upload new recipes with OMERO project/dataset info for each sample.

**Are you prepared to work on it?**
definitely

David-Baddeley, 2020-10-21:
I wonder if this isn't a case where we should just push through with a webby recipe editor, before we hack things too much.

As it stands, the USERFILE parameters are a hacked work-around for the lack of a web-based editor. We could do a similar hack for, e.g. USERSTRING, but that would mean adding more cruft to the recipe classes. Templating anything other than strings gets super hard to do in an intuitive way, and I'd feel uncomfortable with with putting too much work into something which (at least in theory) is a temporary solution.

One advantage the USERFILE templates currently have over a full web recipe editor is that they effectively pull out the critical parameters of a recipe that a user might want to change, so it would be good to think about how to address that.

barentine, 2020-10-21:
Hi David, sorry to be brief I agree that a webby recipe editor would be great, and maybe not too horrific (can honestly be text based, for starters). 

One thing on my mind is that the USERFILE parameters are fine, and USERSTRING similarly easy because they are str, and we can just str comprehend them into our recipe templates. Can do the same for other parameters, but we need to be careful how in order not to lose hard typing. Linking, for completeness/later, our brief discussion on this in #530. 

In my dreams, whatever we do here will:
- allow more than just files to be templated (obviously the main point)
- allow easy rule-to-rule linking in PYMEAcquire, such that a set of chained rules producing one output can have that output used as a file input parameter for recipe modules in another set of chained rules. 
- allow the same yaml to work for both cluster, templated, and bakeshop/dh5view/pymevis. This one is probably a stretch, but is along the lines of my hopes for #464 that we can maybe keep moving towards minimal changes if any when trying to accomplish the same task with the same recipe.

David-Baddeley, 2020-10-21:
[image]

Just remembered that we already had some initial attempts at a webby recipe editor & added a small amount of CSS ... there's still a bunch to do, but I don't think it would be too much work to get it up and running.

We should probably firm up what is the use case for templated recipes, and specifically what they add over the open, modify, use modified recipe use case. To the points above:

- Is there a compelling reason not to just edit the recipe parameters? We already have editors which let you edit the parameters without changing connectivity ... 
- rule linking is already handled effectively with the templated output names that are in recipes at the moment (i.e. you should be able to make one substitution with, e.g. `file_stub` right the way through a rule chain and get linked rules that work. Requires you to actually look at what the outputs of the previous recipe are when you are putting them together, but I don't think that this is unreasonable (we could potentially add some sanity checking later)
- I'm not sure that adding more templating helps here ... .

As I see it the biggest argument for templating is exposing only a small subset of the recipes parameters, but I wonder if there is a more robust way of doing this. I'm really worried that we are piling abstraction on top of abstraction on top of abstraction, which leads to a complex (and probably fragile) edifice.

barentine, 2020-10-21:
web gui fixes a lot of things, I'd forgotten about `PYME.recipes.html_recipe_editor.py`. I think that would solve this current github issue completely.

The templated is nice, but we could unify the web views on the cluster (rather than templated, standalone, now editable) and the recipe yamls if we checked parameter type to know e.g. which module input editors trigger the js file selector menu

Lingering, and relevant as a potential reason not to just edit recipe parameters / related to #528 , is I want to queue 96 tile protocols, each of which detects/queues another protocol at many ROI, and I want all of the ROI series to be spooled to a directory on the cluster corresponding to their well, and uploaded to an OMERO dataset corresponding to their well. That seems accomplishable to me with templated arguments, but not practically solved with the 'edit each time' approach.

David-Baddeley, 2020-10-21:
I think the #528 stuff should be able to be captured with the current templating of output names - we might just need to expand the context a bit / let rules include a few context variables for the recipe outputs. I'd be keen to keep this to the output modules and where files go, rather then a broader recipe templating if possible.

barentine, 2020-11-25:
closing in favor of the html recipe editor

closed 2020-11-25 by barentine