python-microscopy/python-microscopy #475: [BUG] Can't open h5 in PYMEVis

zacsimile, 2020-10-06:
**Describe the bug**
```
 Error whilst running File>&Open
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 736, in OpenFile
    self.pipeline.OpenFile(filename, **args)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 778, in OpenFile
    ds = self._ds_from_file(fn, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 680, in _ds_from_file
    self.recipe._inject_tables_from_h5('', h5f, filename, '.hdf')
AttributeError: 'ModuleCollection' object has no attribute '_inject_tables_from_h5'

```

**To Reproduce**
Steps to reproduce the behavior:
1. Open an `h5` file in `PYMEVis`

**Expected behavior**
The file opens

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
Add any other context about the problem here.

zacsimile, 2020-10-06:
Following #477, I still can't open `h5` files but I get a different error.

```
 Error whilst running File>&Open
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 736, in OpenFile
    self.pipeline.OpenFile(filename, **args)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 778, in OpenFile
    ds = self._ds_from_file(fn, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 680, in _ds_from_file
    self.recipe._inject_tables_from_hdf5('', h5f, filename, '.hdf')
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 936, in _inject_tables_from_hdf5
    if t.name == 'Events':
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/tables/group.py", line 836, in __getattr__
    return self._f_get_child(name)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/tables/group.py", line 708, in _f_get_child
    self._g_check_has_child(childname)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/tables/group.py", line 395, in _g_check_has_child
    % (self._v_pathname, name))
tables.exceptions.NoSuchNodeError: group ``/MetaData`` does not have a child named ``name``
```

zacsimile, 2020-10-06:
Tried changing line 936 of `PYME.recipes.base.py` from `if  t.name == 'Events':` to `if hasattr(t, 'name') and (t.name == 'Events'):`. This results in the following error, which makes me think it's not quite that simple to fix.

```
 Error whilst running File>&Open
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 167, in __getattr__
    return self[item]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 468, in __getitem__
    raise KeyError('Key (%s) not found' % key)
KeyError: 'Key (resultsSource) not found'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/visCore.py", line 736, in OpenFile
    self.pipeline.OpenFile(filename, **args)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 778, in OpenFile
    ds = self._ds_from_file(fn, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 685, in _ds_from_file
    ds = ds_.resultsSource
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 169, in __getattr__
    raise AttributeError("'%s' has not attribute '%s'" % (self.__class__, item))
AttributeError: '<class 'PYME.IO.tabular.HDFSource'>' has not attribute 'resultsSource'
```

I don't have a good grasp on the new loading architecture yet. Any ideas @barentine?

barentine, 2020-10-06:
I haven't played the new loading after #284 went in with tweaks/on top of other events stuff. Will take a look

barentine, 2020-10-06:
might just need to check `t` is a `tables.VLArray` before any `t.name` business so we don't hit it on the metadata group

David-Baddeley, 2020-10-06:
The resultsSource bit is my fault - got a bit too clever when tweaking the
PR. Basically the tables were previously wrapped in a mapping filter, now
directly injected.

David-Baddeley, 2020-10-06:
will fix the resultsSource bit in a sec.

David-Baddeley, 2020-10-07:
should be closed now, added a new test (#488) to hopefully prevent such issues in the future

closed 2020-10-07 by David-Baddeley