python-microscopy/python-microscopy #478: [BUG] meshing compilation fails in some environments

barentine, 2020-10-06:
**Describe the bug**
```
PYME/experimental/_triangle_mesh.c:11004:3: warning: passing argument 4 of ‘_update_face_normals’ from incompatible pointer type [[0/327]
 by default]
   _update_face_normals((&(*((__pyx_t_5numpy_int32_t *) ( /* dim=0 */ (__pyx_v_idxs.data + __pyx_t_7 * __pyx_v_idxs.strides[0]) )))), __$
yx_v_self->_chalfedges, __pyx_v_self->_cvertices, __pyx_v_self->_cfaces, (__pyx_v_idxs.shape[0]));
   ^
In file included from PYME/experimental/_triangle_mesh.c:633:0:
PYME/experimental/triangle_mesh_utils.c:316:13: note: expected ‘struct face_t *’ but argument is of type ‘struct face_d *’
 static void _update_face_normals(int32_t *f_idxs, halfedge_t *halfedges, vertex_t *vertices, face_t *faces, signed int n_idxs)
             ^
error: Command "gcc -pthread -B /home/ubuntu/miniconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Ws$
rict-prototypes -fPIC -IPYME/experimental -I/home/ubuntu/miniconda3/lib/python3.6/site-packages/numpy/core/include -I/home/ubuntu/minicon
da3/lib/python3.6/site-packages/numpy/core/include -I/home/ubuntu/miniconda3/include/python3.6m -I/home/ubuntu/miniconda3/include/python3
.6m -c PYME/experimental/_triangle_mesh.c -o build/temp.linux-x86_64-3.6/PYME/experimental/_triangle_mesh.o -MMD -MF build/temp.linux-x86
_64-3.6/PYME/experimental/_triangle_mesh.o.d -O3 -fno-exceptions -ffast-math -march=native -mtune=native" failed with exit status 1

```
<details> <summary> on this build </summary>

```
# packages in environment at /home/ubuntu/miniconda3:
#
# Name                    Version                   Build  Channel
absl-py                   0.9.0                    pypi_0    pypi
alabaster                 0.7.12                   py36_0    anaconda
appdirs                   1.4.3                    pypi_0    pypi
asn1crypto                1.3.0                    py36_0    anaconda
astor                     0.8.1                    pypi_0    pypi
attrs                     19.3.0                     py_0    anaconda
babel                     2.8.0                      py_0    anaconda
beautifulsoup4            4.9.1                    py36_0    anaconda
blas                      1.0                         mkl    anaconda
brotlipy                  0.7.0           py36h7b6447c_1000    anaconda
bzip2                     1.0.8                h7b6447c_0    anaconda
ca-certificates           2020.7.22                     0    anaconda
cairo                     1.14.12              h8948797_3    anaconda
certifi                   2020.6.20                py36_0    anaconda
cffi                      1.14.0           py36h2e261b9_0    anaconda
chardet                   3.0.4                 py36_1003    anaconda
cheroot                   8.3.0                    py36_0    anaconda
cherrypy                  18.1.1                   py36_0    anaconda
cloudpickle               1.5.0                      py_0    anaconda
conda                     4.8.4                    py36_0    anaconda
conda-build               3.18.11                  py36_0    anaconda
conda-package-handling    1.6.1            py36h7b6447c_0    anaconda
contextlib2               0.6.0.post1                py_0    anaconda
cryptography              2.9.2            py36h1ba5d50_0    anaconda
cycler                    0.10.0                   py36_0    anaconda
cython                    0.29.13          py36he6710b0_0    anaconda
cytoolz                   0.10.1           py36h7b6447c_0    anaconda
dask-core                 2.20.0                     py_0    anaconda
dbus                      1.13.12              h746ee38_0    anaconda
decorator                 4.4.2                      py_0    anaconda
dispatch                  1.0                      py36_0    david_baddeley
docutils                  0.16                     py36_1    anaconda
expat                     2.2.9                he6710b0_2    anaconda
fftw                      3.3.8                h7b6447c_3    anaconda
filelock                  3.0.12                     py_0    anaconda
fontconfig                2.13.0               h9420a91_0    anaconda
freetype                  2.10.2               h5ab3b9f_0    anaconda
fribidi                   1.0.9                h7b6447c_0    anaconda
future                    0.18.2                   py36_1    anaconda
gast                      0.3.3                    pypi_0    pypi
gettext                   0.19.8.1             h9b4dc7a_1    anaconda
glib                      2.56.2               hd408876_0    anaconda
glob2                     0.7                        py_0    anaconda
graphite2                 1.3.14               h23475e2_0    anaconda
grpcio                    1.31.0                   pypi_0    pypi
gst-plugins-base          1.14.0               hbbd80ab_1    anaconda
[... 318 lines cut ...]
yaml                      0.1.7                h96e3832_1    anaconda
zc-lockfile               2.0                      pypi_0    pypi
zeroc-ice36-python        3.6.5            py37h6bb024c_6    ome
zeroconf                  0.17.7                   pypi_0    pypi
zipp                      3.1.0                      py_0    anaconda
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.4                h0b5b093_3    anaconda

```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. `python setup.py develop` in the 'wrong' environment

**Expected behavior**
egg link code without bricking

 - OS: [e.g. Win10] ubuntu 14.4 (fail)/ 18.04 (ok)
 - Python version: 3.6 (fail) / 3.7 (ok)
 - numpy version: 1.14 (fail) / 1.19 (ok)

**Additional context**
without looking, maybe related to discussion around #451 #453

David-Baddeley, 2020-10-06:
From the environments, I suspect it has something to do with the cython version and the c code that cython is generating. It could also be something in the interaction between cython and numpy. 

It looks like cython versions 0.29.13 and 0.29.21 respectively. Might be worth looking at the changelog between these versions (or at the very least seeing if we can trigger by just changing cython version rather than the full environment)

barentine, 2020-10-06:
can try updating cython to 0.29.21 shortly

barentine, 2020-10-07:
[image]
cython version alone isn't enough.

barentine, 2020-10-08:
For reference, gcc versions are 4.8.4 (not compiling) and 7.5.0 (compiling). Not that it will matter after #489

barentine, 2020-10-09:
Still failing despite #489 @David-Baddeley and @zacsimile 
[image]

zacsimile, 2020-10-09:
This seems like a separate error from what was fixed in #489. This code was written about a year ago, so I'm surprised you haven't run into this compile bug before. Is there anything else new, package-wise?

zacsimile, 2020-10-09:
O sorry, I was looking at the warning. The error appears to come from the `int` declaration in the `for` loop. Will upload a fix.

barentine, 2020-10-13:
solved with #503 and #489. Thanks @David-Baddeley and @zacsimile

closed 2020-10-13 by barentine