python-microscopy/python-microscopy #481: [BUG] PYMEVis Help>About displays nothing and throws an error

zacsimile, 2020-10-06:
**Describe the bug**
```
 Error whilst running Help>&About
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
TypeError: 'NoneType' object is not callable
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open PYMEVis
2. Click `Help>About`

**Expected behavior**
An about window pops up.

**Possible alternative**
Remove `Help>About`.

David-Baddeley, 2020-10-07:
That said, the current 'About' dialog is super out of date and useless. Might almost be better to remove the menu item completely.

a) Do we actually need/want a `help-->about` menu item? What it actually bring us? 
b) If we do, what info should we present in `help-->about` and should we have one about dialog which is re-used across all programs.

David-Baddeley, 2020-10-07:
Seeing as the bug is now fixed, changing this to a discussion issue about what help-->about should actually show.

zacsimile, 2020-11-05:
I think it should include the PYME version number and maybe a link to our issue tracker labeled `Report an issue`.

zacsimile, 2020-12-04:
Do we want to continue having this discussion, or are we satisfied with the current about dialog?

csoeller, 2020-12-04:
One bit of info it should show in addition to what is there now is the supporting packages version set etc as at the top of bug reports:

```
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

```
I am always looking for this info when I want to report a bug that did not throw open the 'bug window'. Tedious to do it manually.

barentine, 2020-12-04:
I agree with @csoeller on this one. I never learned the windows equivalent of `grep`, so when I'm troubleshooting on someone elses machine a quick 'survey of the environment' in PYME would be grand