python-microscopy/python-microscopy #483: [ENHANCEMENT] Add Help>Update to PYMEVis, PYMEImage, PYMEAcquire

zacsimile, 2020-10-06:
**Is your feature request related to a problem? Please describe.**
It would be nice to add a graphical method of updating PYME. Asking users to open `Anaconda prompt` and type `conda update python-microscopy` is a bit much for folks who installed using single-click.

**Describe the solution you'd like**
Add a `Help>Update` menu item that calls `conda update python-microscopy` for the user, and maybe prompts the user to relaunch on success.

**Describe alternatives you've considered**
Do nothing.

**Are you prepared to work on it?**
Yep

zacsimile, 2020-10-06:
I'm not sure of the best way to implement this. The quick and dirty way is 

```
import subprocess
ret = subprocess.call(‘conda update python-microscopy -y', shell=True)
if ret == 0:
    # success, offer a restart
```

The problem is, this will be successful even if the update wasn't necessary. Ideally we'd compare versions first. Shockingly, I can not find an easy way to just check if an update exists for a `conda` package without doing anything else. Thoughts?

David-Baddeley, 2020-10-06:
The solution I had in mind is as follows: 
- put a file (or potentially CGI endpoint) on python-microscopy.org - e.g. current_version.txt. 
- update this whenever we do a release
- client applications fetch this on startup and compare contents to PYME.version, potentially prompting for an upgrade

The advantage of this approach is that it gives us usage metrics (we get to count the number of requests for that file on the server - potentially also sorted by unique IP) for grants etc without adding too much intrusive telemetry. There's an uncommitted (or potentially in PYMEnf) `pyme_phone_home.py` which has a very rough start to this.

David-Baddeley, 2020-10-06:
There are a few wrinkles in actually implementing the upgrade as we have to detect if they are using a conda-, pip, or source install, so as a first step I'd probably just display a message box indicating that an update was available.

zacsimile, 2020-10-06:
I'm not sure if we do need to make those detections. I think it might be safe to say the update only works for packaged installs as a first pass.

How do we check if an update is available? Do we compare `PYME.version.version` to the `conda` channel? PyPI? GitHub?

zacsimile, 2020-10-06:
I suppose `git` is the obvious choice since we trigger builds for pip and conda from here.

zacsimile, 2020-10-06:
OK, here's how I propose doing this, following https://docs.github.com/en/free-pro-team@latest/rest/reference/repos#releases.

```
import requests
import PYME.version
r = requests.get('https://api.github.com/repos/python-microscopy/python-microscopy/releases/latest')
if PYME.version.version < r.json()['tag_name']:
    # time to upgrade
```

David-Baddeley, 2020-10-06:
As stated above, we compare `PYME.version.version` with `requests.get('https:\\www.python-microscopy.org\current_version')`

Requires us to update the `current_version` file every time we build packages, but that should be very doable.

zacsimile, 2020-10-06:
Hmm, for some reason I didn't see your first post. What is the advantage to this over using GitHub to automatically track releases?

David-Baddeley, 2020-10-06:
We get metrics we can use in grants ...

Ultimately I'd like it to be something like:
requests.get('https:\\www.python-microscopy.org\current_version?os=XXX&python=XXX&called_as=`sys.argv`&mac=XX.XX.XX.XX&installed_version=XXX')

But we'd have to think a bit about the security/privacy implications and at the very least give users a GUI route to opt out. Might be best to defer the enhanced metrics until after we have a GUI config editor.

zacsimile, 2020-10-06:
Yeah, sorry. I edited my response I think before you got here 'cause I re-read your first post and noticed. 

I still think that we need a GUI upgrade. Just checking for updates isn't all that helpful because I think most of the people who can do a command line update already know how to look for a new version. If we just do a GUI "Update Available," we're not solving anyone's problem.

I suppose we do still get metrics for grants...

David-Baddeley, 2020-10-06:
I'm not arguing against a GUI update route. I also think it's important, but should realistically be step 2. It's potentially going to have to be a standalone updater rather than a menu on any of the existing programs as we need to overwrite .dll/.so (s) which doesn't work on windows or linux if they are in use (i.e. we need to kill all open PYME programs which use c extension modules before doing the update).

David-Baddeley, 2020-10-06:
*curiously over-writing .so on the fly does seem to work on the mac

zacsimile, 2020-10-06:
OK. Sounds good to me! I like step 1.

barentine, 2020-10-30:
just going to link this back to #416 as backend for the conda route (which as I understand is probably the biggest bang for the buck)

zacsimile, 2021-03-12:
Gonna bump this again. I feel like a little sting every time a program automatically provides an alert to update and then automatically updates itself when I press "Update Now". This happens with every other program I use, without exception.

David-Baddeley, 2022-11-29:
I think we can close this now ...

closed 2022-11-29 by David-Baddeley