python-microscopy/python-microscopy #490: [BUG] RuleServer counting has potential to get bad if tasks fail/are reassigned

barentine, 2020-10-08:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. run a recipe or localization with a couple tasks which will fail
2. wait long enough for the failed tasks to get reassigned
3. if you're sufficiently unlucky you'll see some bad counting

**Expected behavior**
maybe not automatically requeue failed tasks - just let them fail and maybe we add an interface for quickly requeueing them from clusterUI if a user would like. That said, if we keep the requeue stuff, we should also sort out the counting - probably a locking issue.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] ubuntu 14.04/18.04
 - Python version: 3.6 / 3.7
 - numpy version: 1.14 / 1.19

David-Baddeley, 2020-10-15:
Bug is in `IntegerIDRule.mark_complete()` and is that we don't check what the current status is before updating `nCompleted`, `nFailed`, and `nAssigned`. This gets triggered by the following flow:

- send out a task
- task times out and is re-queued 
- original task is completed and returned (decrementing nAssigned, despite the fact that it has been re-assigned)
- requeued task is completed and returned (decrementing nAssigned again)

This is a symptom of the timeout being too short. With current code we at least get some indication that something is going wrong. Dropping the retries just masks this.

There are various nasty corollaries where the behaviour will result in truncated analysis (we call the rule done when `nAssigned <1`). I'll try and fix the accounting so that it reports things more reasonably - we might need to add a field or two to a) report how many timed out and b) how many returned after timeout.

barentine, 2020-10-15:
Sounds good. Thanks @David-Baddeley

barentine, 2020-10-16:
I ran the updated ruleserver last night with some templated from-cluster recipes using inputsByTask. Will note :  
- I am suspicious we are ~assigning an entire rule to a single nodeserver rather than distributing recipe tasks more like localization tasks (was suspicious of this a couple weeks ago, didn't immediately see why that was happening when i looked at the code). Maybe we make a separate issue on that.
- those tasks run long, and we don't currently have a GUI-way of setting the task/rule timeout anywhere. I should file an enhancement on this, but meanwhile I bushwhacked and make the task timeout the same as the rule timeout (3600 s). What I got was duplicated results, i.e. they ran twice. If I recall from prior discussions while looking over the ruleserver, we don't think there's an easy way to check if the task has already been returns without turfing performance during e.g. localization. I haven't looked again, but is that still the case? When I upload the same things twice to a server I'm not super worried about it, but for e.g. `pyme-cluster:// - aggregate` someone who doesn't realize their rule silently timed out but still reported might not catch they duplicated a bunch of stuff - especially if they have any stochastic e.g. monte-carlo/DBSCAN  stuff going on in their recipe. At least a simple enhancement request I'll file related to this is to maybe write some rule metadata into results files? It's kind of crazy/haven't thought about how it might work, but at least for localization results it wouldn't hurt to see in the results metadata whether you lost a frame (or 20) because you were imaging live/localizing at full frame rate while Andrew was in the other room queuing a hundred 10 min recipe tasks.

barentine, 2020-10-23:
[image]

barentine, 2020-10-23:
Corresponding log (for -30 failed) appears not to exist
```
Traceback:

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/home/smeagol/code/python-microscopy/PYME/cluster/clusterUI/clusterbrowser/views.py" in file
  16.         return HttpResponse(clusterIO.get_file(filename, use_file_cache=False), content_type='')

File "/home/smeagol/code/python-microscopy/PYME/IO/clusterIO.py" in get_file
  806.         raise IOError("Specified file could not be found: %s" % filename)

Exception Type: OSError at /files/download/LOGS/rules/000024-e836de989f254c8dba4ddcf421f2cc6b.log
Exception Value: Specified file could not be found: LOGS/rules/000024-e836de989f254c8dba4ddcf421f2cc6b.log
```

barentine, 2020-11-14:
maybe needs a bit more attention
[image]

David-Baddeley, 2022-12-14:
I think this is finally fixed with #1313

closed 2022-12-14 by David-Baddeley