python-microscopy/python-microscopy #492: [BUG] pipelineify in pipeline recipe hiccup: metadata writing + file opened as read

barentine, 2020-10-08:
**Describe the bug**
New pipelineify in the pipeline recipe is something I'm pretty excited about, but we might need to wrap metadata handlers in nestedclassmetadatahandlers in recipe modules which add metadata (ideally all of them) or on load, to avoid the following
```
 Error whilst running Corrections>Multiview>Shift correct folded channels
==============================================
python-microscopy=20.10.05
python=3.6.9, platform=linux
numpy=1.16.5, wx=4.0.4 gtk2 (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 898, in _create_path
    child = parent._f_get_child(pcomp)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/group.py", line 708, in _f_get_child
    self._g_check_has_child(childname)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/group.py", line 395, in _g_check_has_child
    % (self._v_pathname, name))
tables.exceptions.NoSuchNodeError: group ``/MetaData/Multiview`` does not have a child named ``shift_map``

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/LMVis/Extras/multiviewMapping.py", line 165, in OnShiftCorrectFolded
    recipe.execute()
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/multiview.py", line 92, in execute
    mapped.mdh['Multiview.shift_map.location'] = loc
  File "/home/smeagol/code/python-microscopy/PYME/IO/MetaDataHandler.py", line 281, in __setitem__
    self.setEntry(name, value)
  File "/home/smeagol/code/python-microscopy/PYME/IO/MetaDataHandler.py", line 471, in setEntry
    currGroup = self.h5file._get_or_create_path('/'.join(['', 'MetaData']+ ep), True)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 877, in _get_or_create_path
    return self._create_path(path)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 900, in _create_path
    child = create_group(parent, pcomp)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 940, in create_group
    title=title, new=True, filters=filters)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/group.py", line 235, in __init__
    super(Group, self).__init__(parentnode, name, _log)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/node.py", line 249, in __init__
    file_._check_writable()
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 2167, in _check_writable
    raise FileModeError("the file is not writable")
tables.exceptions.FileModeError: the file is not writable

```

slightly more generally
```
>>> pipeline.mdh['tesssst'] = 5
>>> type(pipeline.mdh)
<class 'PYME.IO.MetaDataHandler.NestedClassMDHandler'>
>>> pipeline.selectedDataSource
<PYME.IO.tabular.MappingFilter object at 0x7f88a53ffc88>
>>> pipeline.selectedDataSource.mdh
1601329119.6013823
0.0012570225563909773
0.0012570225563909773
<HDFMDHandler>:

ActiveCamera: HamamatsuORCA
EstimatedLaserOnFrameNo: 32
StartTime: 1601329119.6013823 (2020-09-28 17:38:39)
imageID: 0
Analysis.BGRange: [-32, 0]
Analysis.DebounceRadius: 4
Analysis.DetectionFilterSize: 4
Analysis.DetectionThreshold: 1.1
Analysis.FiducialThreshold: 1.8
Analysis.FitModule: AstigGaussGPUFitFR
Analysis.GPUPCTBackground: True
Analysis.PCTBackground: 0.25
Analysis.ROISize: 7.5
Analysis.StartAt: 32
Analysis.TrackFiducials: False
Analysis.subtractBackground: True
Camera.ADOffset: 101.753685
Camera.Binning: [1, 1]
Camera.CycleTime: 0.0012570225563909773
Camera.DarkMapID: 
Camera.EMGain: 1
Camera.ElectronsPerCount: 0.416613
Camera.FlatfieldMapID: 
Camera.IntegrationTime: 0.0012570225563909773
Camera.Model: C11440-22C
Camera.Name: Hamamatsu ORCA Flash 4.0
Camera.NoiseFactor: 1
Camera.ROI: [180, 896, 1940, 1152]
Camera.ROIHeight: 256
Camera.ROIOriginX: 180
Camera.ROIOriginY: 896
Camera.ROIWidth: 1760
Camera.ReadNoise: 0.997
Camera.SensorHeight: 2048
Camera.SensorWidth: 2048
Camera.SerialNumber: 720795
Camera.StartCCDTemp: 0
Camera.TrueEMGain: 1
Camera.VarianceMapID: 
Camera.Views: [0, 1, 2, 3]
CameraOrientation.FlipX: False
CameraOrientation.FlipY: False
CameraOrientation.Rotate: False
Multiview.ActiveViews: [0, 1, 2, 3]
Multiview.ChannelColor: [0, 1, 1, 0]
Multiview.NumROIs: 4
Multiview.ROI0Origin: [180, 896]
Multiview.ROI1Origin: [744, 896]
Multiview.ROI2Origin: [1144, 896]
Multiview.ROI3Origin: [1684, 896]
Multiview.ROISize: [256, 256]
Positioning.x: 427.2
Positioning.x_piezo: 427.2
Positioning.x_target: 427.2
Positioning.y: -3656.2
Positioning.y_piezo: 3656.2
Positioning.y_target: -3656.2
Positioning.z: 43.949
Positioning.z_piezo: 43.9484
Positioning.z_target: 43.96
Protocol.DataStartsAt: 0
Protocol.Filename: htsms-staggered.py
Protocol.PiezoStartPos: 43.96
Protocol.ZStack: True
StackSettings.EndPos: 50.75749999999999
StackSettings.NumSlices: 12
StackSettings.ScanMode: Start and End
StackSettings.ScanPiezo: z
StackSettings.StartPos: 43.94839999999999
StackSettings.StepSize: 0.6
voxelsize.units: um
voxelsize.x: 0.115
voxelsize.y: 0.115
voxelsize.z: 0.6
Lasers.MPB560.AOTFSetting: 17.996534996534997
Lasers.MPB560.LaserPower: 2199.661590524535
Lasers.MPB560.MaxPower: 650.1
Lasers.MPB560.MinPower: 0.0
Lasers.MPB560.On: False
Lasers.MPB560.Power: 575.0
Lasers.MPB642.AOTFSetting: 19.19257993035771
Lasers.MPB642.LaserPower: 2198.697068403909
Lasers.MPB642.MaxPower: 675.4
Lasers.MPB642.MinPower: 0.0
Lasers.MPB642.On: False
Lasers.MPB642.Power: 578.0
Lasers.OBIS405.AOTFSetting: 0.0
Lasers.OBIS405.LaserPower: 100.0
Lasers.OBIS405.MaxPower: 1.1
Lasers.OBIS405.MinPower: 0.0
Lasers.OBIS405.On: False
Lasers.OBIS405.Power: 0.0
Lasers.OBIS488.AOTFSetting: 0.0
Lasers.OBIS488.LaserPower: 51.724137931034484
Lasers.OBIS488.MaxPower: 5.316666666666666
Lasers.OBIS488.MinPower: 0.0
Lasers.OBIS488.On: False
Lasers.OBIS488.Power: 0.0
>>> type(pipeline.selectedDataSource.mdh)
<class 'PYME.IO.MetaDataHandler.HDFMDHandler'>
>>> pipeline.selectedDataSource.mdh['test'] = 5
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/smeagol/code/python-microscopy/PYME/IO/MetaDataHandler.py", line 281, in __setitem__
    self.setEntry(name, value)
  File "/home/smeagol/code/python-microscopy/PYME/IO/MetaDataHandler.py", line 472, in setEntry
    currGroup._f_setattr(en, value)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/node.py", line 896, in _f_setattr
    setattr(self._v_attrs, name, value)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/attributeset.py", line 466, in __setattr__
    nodefile._check_writable()
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/tables/file.py", line 2167, in _check_writable
    raise FileModeError("the file is not writable")
tables.exceptions.FileModeError: the file is not writable
```

**To Reproduce**
Steps to reproduce the behavior:
1. open an h5r file in PYMEVis
2. try and add something to the selectedDataSource.mdh rather than the pipeline.mdh
4. See error

David-Baddeley, 2020-10-09:
In general I can't think of a situation where adding something to `selectedDataSource.mdh` (or even pipeline.mdh) is a good idea so it's not completely unreasonable to throw an error. Have merged #495 as I'm more worried that we might at some point get a file that wasn't loaded read only. We might want to revisit at some point and make an explicitly read-only metadatahandler and use that, but that's probably not urgent.

barentine, 2020-10-13:
fixed in #500 , and more permanent in #502

closed 2020-10-13 by barentine