python-microscopy/python-microscopy #507: [BUG] Adding a recipe tile with bad input/outputs spec'd doesn't give you a chance to fix it

barentine, 2020-10-12:
**Describe the bug**
We could stand to make the recipe GUI a bit kinder, e.g. raise dialog and cancel the add when this happens on Add Module:
```
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 196, in process
    func(*args, **kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeGui.py", line 531, in OnPick
    self.configureModule(k)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeGui.py", line 553, in configureModule
    k.edit_no_invalidate(handler=MControl())
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 264, in edit_no_invalidate
    self.invalidate_parent()
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 104, in invalidate_parent
    self._parent.prune_dependencies_from_namespace(self.outputs)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 571, in prune_dependencies_from_namespace
    downstream = list(keys_to_prune) + list(self._getAllDownstream(rdg, list(keys_to_prune)))
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 560, in _getAllDownstream
    downstream.update(self._getAllDownstream(rdg, list(next_level)))
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 560, in _getAllDownstream
    downstream.update(self._getAllDownstream(rdg, list(next_level)))
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 560, in _getAllDownstream
    downstream.update(self._getAllDownstream(rdg, list(next_level)))
  [Previous line repeated 970 more times]
  File "/home/smeagol/code/python-microscopy/PYME/recipes/base.py", line 546, in _getAllDownstream
    downstream = set()
RecursionError: maximum recursion depth exceeded while calling a Python object
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/wx/core.py", line 2228, in Notify
    self.notify()
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/wx/core.py", line 3384, in Notify
    self.result = self.callable(*self.args, **self.kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeGui.py", line 489, in update
    self.recipePlot.draw()
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeGui.py", line 77, in draw
    node_positions, connecting_lines = recipeLayout.layout(dg)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeLayout.py", line 37, in layout
    ips, yvs = depGraph.arrangeNodes(dg)
  File "/home/smeagol/code/python-microscopy/PYME/misc/depGraph.py", line 35, in arrangeNodes
    ts = list(toposort.toposort(dg))
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/toposort.py", line 71, in toposort
    raise ValueError('Cyclic dependencies exist among these items: {}'.format(', '.join(repr(x) for x in data.items())))
ValueError: Cyclic dependencies exist among these items: ('colour_mapped', {<PYME.recipes.localisations.ProcessColour object at 0x7f1fe5d99678>}), (<PYME.recipes.tablefilters.FilterTable object at 0x7f1fc97223b8>, {'colour_mapped'}), ('filtered_localizations', {<PYME.recipes.tablefilters.FilterTable object at 0x7f1fc97223b8>}), (<PYME.recipes.tablefilters.FilterTable object at 0x7f1fc9722468>, {'filtered_localizations'}), ('filtered_input', {<PYME.recipes.tablefilters.FilterTable object at 0x7f1fc9722468>}), (<PYME.recipes.multiview.Fold object at 0x7f1fc9722518>, {'filtered_input'}), ('folded', {<PYME.recipes.multiview.Fold object at 0x7f1fc9722518>}), (<PYME.recipes.multiview.ShiftCorrect object at 0x7f1fc9722570>, {'folded'}), ('shift_corrected', {<PYME.recipes.multiview.ShiftCorrect object at 0x7f1fc9722570>}), (<PYME.recipes.multiview.FindClumps object at 0x7f1fc97225c8>, {'shift_corrected'}), ('with_clumps', {<PYME.recipes.multiview.FindClumps object at 0x7f1fc97225c8>}), (<PYME.recipes.multiview.MergeClumps object at 0x7f1fc9722620>, {'with_clumps'}), ('clumped', {<PYME.recipes.multiview.MergeClumps object at 0x7f1fc9722620>}), (<PYME.recipes.multiview.MapAstigZ object at 0x7f1fe6a75308>, {'clumped'}), ('z_mapped', {<PYME.recipes.multiview.MapAstigZ object at 0x7f1fe6a75308>}), (<PYME.recipes.localisations.ProcessColour object at 0x7f1fe5d99678>, {'z_mapped'})

```

**To Reproduce**
Steps to reproduce the behavior:
1. create a cyclic dependency
2. recipeGUI will throw/freeze up. You can e.g. `clear recipe` but it won't clear and you'll get a similar traceback 

**Expected behavior**
raise a dialog and say 'bad choice, fix it'

David-Baddeley, 2020-10-12:
Yep, cyclic dependencies are an issue. Suspect it has got worse with the updating fix. Workaround used to be to edit the recipe text and apply. Does this still work?

Ideally we'd catch this at the editor level (maybe adding trait validation of some sort ) and never let people create cyclic dependencies.

David-Baddeley, 2020-11-27:
Current status:

- with merge of #634, you can no longer create cyclic dependencies with the graphical editor
- you can still create them in the text editor / or by loading a bad recipe
- once created, they are not recoverable (require restart)

barentine, 2020-12-16:
going to call this good enough for now. Thanks David

closed 2020-12-16 by barentine