python-microscopy/python-microscopy #510: Visgui is unable to 'open' localization files from splitterfitter analysis

lukasfue, 2020-10-12:
**Describe the bug**
Visgui is unable to 'open' localization files from splitterfitter analysis. Files I have definitely opened in that past are unable to be opened now.

**To Reproduce**
Steps to reproduce the behavior:
1. File -> Open -> Choose either an H5R or HDF localization file that was created using SplitterFitterFNR analysis on 2-color ratiometric data.
2. See error

Error whilst running File>&Open
==============================================
python-microscopy=20.10.05
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\visCore.py", line 736, in OpenFile
    self.pipeline.OpenFile(filename, **args)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\pipeline.py", line 823, in OpenFile
    self.recipe.execute()
  File "c:\users\laf62\python-microscopy\PYME\recipes\base.py", line 641, in execute
    m.execute(self.namespace)
  File "c:\users\laf62\python-microscopy\PYME\recipes\localisations.py", line 206, in execute
    mapped_ds.addColumn('nPhotonsg', pipeline.getPhotonNums({'A': mapped_ds['fitResults_Ag'], 'sig': mapped_ds['fitResults_sigma']}, self.mdh))
AttributeError: 'Pipelineify' object has no attribute 'mdh'

**Expected behavior**
Files open normally.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.8.3
 - numpy version: 1.19.1

barentine, 2020-10-12:
Thanks @lukasfue and sorry about that!

barentine, 2020-10-13:
@lukasfue  and @David-Baddeley would one of you mind closing this?

closed 2020-10-19 by David-Baddeley