python-microscopy/python-microscopy #516: [ENHANCEMENT] adjusting PYMEAcquire acq tasks/analysis tasks/gui in general + plug-able menu items

barentine, 2020-10-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I never use the zstack dialog, nor do I use the localization settings within hdfspoolframe. Nor do I need either of the tiling panels to be panels, but seem to remember we don't have plug-able menu items in PYMEAcquire at the moment.
[image]
Others have various things they'd like displayed (though in currently existing panels) e.g. #425

**Describe the solution you'd like**
We've had some discussion about 'skinning' PYMEAcquire, the web GUI, etc. . These things are all great, but I think we can kick something in quickly, e.g. #373 or pluggable menu items, or both, or ___, which would make things easier for some of us (maybe just me, but I'm ok with that ;)  ). Alternatively I kick things out of a panel and into another page? Lots of options.

**Describe alternatives you've considered**
waiting

David-Baddeley, 2020-10-14:
Lots of separate points here ... a quick start

- you can already plug menu items in the startup scripts in any of the `@init_gui` sections (the frame derives from `PYME.ui.AUIFrame.AUIFrame` and has an `AddMenuItem()` function. For an example, see the `focusKeys` module (imported in the simulator, and most of my init scripts)

- there is probably a good argument for rationalising the different tiling panels into one tiling panel with, e.g. a dropdown to select the type of tiling.

It might not be that much work to actually get the basic skinning support in (i.e. similar amount of effort but much less broken than the config options). I'm happy to try and find a couple of hours to work on it.

zacsimile, 2021-03-16:
#958, which I recently opened and closed, is an extension of this conversation. @csoeller had some ideas on how to document this properly.