python-microscopy/python-microscopy #524: [ENHANCEMENT] measurement alignment

David-Baddeley, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
There are currently (at least) 4 different ways of measuring cluster/object properties in PYME, each resulting in different column names and interpretations. Each has some functionality that the other lacks. 

These are: 
- `skimage.measure` through `recipes.measurement.Measure2D) (2D, image based)
- `DSView.modules.blobMeasure` (3D, image based)
- `Analysis.points.objectMeasure` (2D, from points)
- `Analysis.points.cluster_morphology` (3D, from points - originally specced as 3D equiv of objectMeasure, but diverged significantly)

**Describe the solution you'd like**
A consistent naming scheme for measurement outputs, and a closer functional alignment. Ideally reducing the number of overlapping implementations.

barentine, 2020-12-02:
Would be nice in this refactor to 
- make `recipes.measurement.Measure2D` multicolor
- make bounding box entries usable (see #103)

David-Baddeley, 2021-07-09:
Just looping back to this re multicolour - I'm a bit wary of making `Measure2D` or it's replacement intrinsically multicolour as it's hard to know, e.g. which labels to use for which channel, and whether there are sensible defaults for this. Although a bit more effort, I think the explicit `SplitChannels`, `Measure2D`, and optionally `AggregateMeasurements` flow makes it very clear, e.g. what mask is being used, and how things are being aligned. I don't see any scenario where it makes sense to have a multi-colour `labels` array, and as most of the measurements are being made on the `labels` array, with comparatively few on the intensity image it would seem odd to have specific multi-channel support for just the intensity data.