python-microscopy/python-microscopy #525: [BUG] Can't re-run a recipe in PYMEVis after it crashes

zacsimile, 2020-10-19:
**Describe the bug**
If I run a recipe with a weird set of parameters and it crashes, e.g. I make a mesh at the wrong density that has no triangles, I get an error traceback in the context manager. If I then try to run the recipe (e.g. meshing) again, but change the parameters to something that works, the recipe still uses the previous, busted parameters. I have to restart `PYMEVis` to change the parameters. This appears to be consistent across recipes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Make a recipe accessed from the PYMEVis menu crash
2. Try to re-run the recipe from the menu with new parameters

**Expected behavior**
The second run should use the updated, rather than the original, parameters. 

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10, macOS 15
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-10-19:
Could you provide a bit more detail (e.g. an example traceback / specific scenario).

In general, if you call something from the menu which adds a recipe module a second time there's a good chance it won't work (it will try and add the recipe module a second time and both modules will be present in the recipe). This is something we should fix - potentially by keeping track of when we add a recipe module in the menu and simply calling `configure_traits()` on the added one rather than trying to re-add - although this is context specific.

The other thing that might be worth considering in the long term is making functions called from the menu transactional (i.e. able to be un-rolled on failure).

zacsimile, 2020-10-19:
Sure! So, I think another common use case might a user accidentally does something that makes no sense, realizes this, and then wants to keep working on their data. For example, if I open a single-channel image and then navigate to `Corrections>Multiview>Fold Channels` I get

```
 Error whilst running Corrections>Multiview>Fold Channels
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/multiviewMapping.py", line 128, in OnFold
    recipe.execute()
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/tablefilters.py", line 38, in execute
    filtered = tabular.ResultsFilter(inp, **self.filters)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 752, in __init__
    raise KeyError('Requested key not present: ' + k)
KeyError: 'Requested key not present: error_y'
```

And then I go whoops, and I decide to make a mesh and I run into the same error again

```
 Error whilst running Mesh>Generate Isosurface
==============================================
python-microscopy=20.10.05
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/extra_layers.py", line 315, in <lambda>
    visFr.AddMenuItem('Mesh', 'Generate Isosurface', lambda e: gen_isosurface(visFr))
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/extra_layers.py", line 88, in gen_isosurface
    recipe.execute()
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/tablefilters.py", line 38, in execute
    filtered = tabular.ResultsFilter(inp, **self.filters)
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 752, in __init__
    raise KeyError('Requested key not present: ' + k)
KeyError: 'Requested key not present: error_y'
```

because my poor choice is still in the recipe

[image]

I think we should at least not add failed recipes to the pipeline (or roll them back) at the time of failure.

David-Baddeley, 2020-10-20:
Have added a transactional `add_modules_and_execute()` to the recipe, and refactored the extras to use it. Try and give this a spin.

zacsimile, 2020-10-20:
Yeah, it works beautifully. Thanks, David! I think a lot of users will love this change.

closed 2020-11-05 by zacsimile