python-microscopy/python-microscopy #539: [BUG] clusterUI Selected Files > Run Recipe does not get any inputs

barentine, 2020-10-25:
**Describe the bug**
When selecting pcs, a recipe, and hitting run recipe
[image]

stdout from clusterUI:
```
PatchedServiceInfo(type='_pyme-taskdist._tcp.local.', name='PYMERuleServer: greenback._pyme-taskdist._tcp.local.', addresses=[b'\n\x96\x07\x84'], port=15346, weight=0, priority=0, server='PYMERuleServer: greenback._pyme-taskdist._tcp.local.', properties={}) b'\n\x96\x07\x84'
intputs by task {}
[25/Oct/2020 17:51:20] "POST /recipes/run/ HTTP/1.1" 302 0
[25/Oct/2020 17:51:20] "GET /status/queues/ HTTP/1.1" 200 6299
[25/Oct/2020 17:51:20] "GET /status/load/ HTTP/1.1" 200 21616
[25/Oct/2020 17:51:20] "GET /status/queues/?_ajax=true HTTP/1.1" 200 2022
[25/Oct/2020 17:51:21] "GET /status/load/ HTTP/1.1" 200 21616
[25/Oct/2020 17:51:21] "GET /s
```

before I reset HTTPRulePusher to HEAD I had some more logging:
```
PatchedServiceInfo(type='_pyme-taskdist._tcp.local.', name='PYMERuleServer: greenback._pyme-taskdist._tcp.local.', addresses=[b'\n\x96\x07\x84'], port=15346, weight=0, priority=0, server='PYMERuleServer: greenback._pyme-taskdist._tcp.local.', properties={}) b'\n\x96\x07\x84'
DEBUG:PYME.cluster.HTTPRulePusher:local_queues: []
INFO:PYME.cluster.HTTPRulePusher:no local rule server, choosing one at random
DEBUG:PYME.cluster.HTTPRulePusher:numTotalFrames = 0
DEBUG:PYME.cluster.HTTPRulePusher:inputs = {'input': []}
intputs by task {}
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] ubuntu 14/16/18
 - Python version: 3.6
 - numpy version: 1.14 / 1.16 / 1.19

David-Baddeley, 2020-10-28:
Potentially fixed with #540

barentine, 2020-10-30:
Yes, this is fixed
```
heheere! filenames: ['pyme-cluster:///Bergamot/2020_9_28/000/analysis/reg.h5r']
[('csrfmiddlewaretoken', ['06LvGJhFxlhTfPUwNzEM3zBzpryHDNBzlGRT1xBlGTPDqA4RBOSH0xduyFnSb2y4']), ('recipeURL', ['RECIPES/calibrate-multiview-shifts.yaml']), ('files', ['pyme-cluster:///Bergamot/2020_9_28/000/analysis/reg.h5r'])]
intputs by task {0: {'input': 'pyme-cluster:///Bergamot/2020_9_28/000/analysis/reg.h5r'}}
```

However, I am still hitting the
```
    self._target(*self._args, **self._kwargs)
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 344, in _poll_tasks
    self._update_tasks()
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 215, in _update_tasks
    taskID, cost = rater.get_local()
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 113, in get_local
    taskID, cost = self.next()
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 107, in next
    return self.__next__()
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 78, in __next__
    task = json.loads(filled_template)
ValueError: Unexpected character in found when decoding object value
```
mentioned in #487