python-microscopy/python-microscopy #543: [BUG] cluster of one logs can be a bit much if the process lingers

barentine, 2020-10-26:
**Describe the bug**
Haven't thought through when/how this happened, e.g. whether I was running some mangled repo that didn't kill these, etc., but we'll want to check whether we're silently doing this
[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
?
**Expected behavior**
don't fill my entire hard drive with logs

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] osx 15
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
I work on PYMECluster stuff and could well have been messing around with things/brought this on myself. Unclear if this effects production installs

David-Baddeley, 2020-10-26:
Could you have a look at what is in those logs? Are there truckloads loads of errors? We could (or potentially should) turn on log rotation, but I don't think the logging is generally shouty enough to get to those sizes in any reasonable time unless something else is going very wrong.

David-Baddeley, 2020-10-26:
actually, just had a look at the logs here - we do log each time we file results  (which means that we get a 125 char line for each frame we analyse) - should add up to a couple of MB per series. We could definitely turn that off, as it's not super informative.

barentine, 2020-10-26:
will try and remember to look when im home. If I recall, we may also log a 'file not found' on every clusterIO.series_exists

barentine, 2020-10-27:
[head500taskWorkerHTTP45179.log](https://github.com/python-microscopy/python-microscopy/files/5445865/head500taskWorkerHTTP45179.log)
[tail500taskWorkerHTTP45179.log](https://github.com/python-microscopy/python-microscopy/files/5445866/tail500taskWorkerHTTP45179.log)

barentine, 2020-10-28:
things to hit:
- file results doesn't need to log every frame / not very useful
- clusterIO.check_exists doesn't

barentine, 2021-03-06:
we might want to limit the logging file size for the taskWorkers. A couple times I've had bad shutdowns on my laptop and wound up with 8x 13 GB log files