python-microscopy/python-microscopy #545: [BUG] cluster get freespace takes 0.45 s every status expiration (1 s)

barentine, 2020-10-27:
**Describe the bug**
Every guiUpdateCallback in `SpoolController` calls the `HDFSpoolFrame` tick which updates freespace. When spooling to cluster, this eventually calls `clusterIO.get_status`. While `clusterIO.get_status` caches, making subsequent calls within the expiration fast, it does not poll updates continuously in the background, meaning that if the cache has expired we call it again, and each time the cache has expired this takes roughly 0.44 s, which is happening in the GUI thread and presumably why I'm seeing the high-throughput system GUI hang every second or so

```
In [7]: import timeit

In [8]: timeit.timeit('get_status("villains")', setup='from PYME.IO.clusterIO import get_status', number=10)
Out[8]: 0.4622703999999942

In [9]: timeit.timeit('get_status("villains")', setup='from PYME.IO.clusterIO import get_status', number=50)
Out[9]: 0.4372077000000161

In [10]: timeit.timeit('get_status("villains")', setup='from PYME.IO.clusterIO import get_status', number=100)
Out[10]: 0.4391475999999841

In [11]:

In [11]: timeit.timeit('get_status("villains"); time.sleep(1)', setup='import time; from PYME.IO.clusterIO import get_s
    ...: tatus', number=10)
Out[11]: 14.45680680000001
```

**To Reproduce**
Steps to reproduce the behavior:
1. run PYMEAcquire with a cluster
2. spool
3. watch the gui stutter a bit

**Expected behavior**
not kill performance for a status update

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] calling from win10
 - Python version: 3.6
 - numpy version: 1.19
 - wxpython version [if a UI bug]: 4.0.4

**Question for @David-Baddeley **
OK if I write `clusterIO.get_polled_status` as a thread-polled version of get_status which starts a thread on its first call and updates every 5 s?

barentine, 2020-10-28:
conversation with David:
- don't put it in clusterIO, maybe config the update poll time
- potentially just skirt the issue with a config option to disable (and update config option docs + cluster_install docs)

follow on: why does it take a long time?

barentine, 2020-10-29:
Sorry if the follow-on should go in this same thread or not, but I'll start it with the `top` during localization analysis of 8x 30k frame series, the GPU usage is substantially lower than the CPU usage, and top on the ruleserver node is:
[image]

barentine, 2020-10-29:
Kicker appears to be a total flop on asynchronous calculation on the gpu buffer. I'll have a look
[20201028-greenback-remFitBuf.pdf](https://github.com/python-microscopy/python-microscopy/files/5456082/20201028-greenback-remFitBuf.pdf)

barentine, 2020-11-01:
Bit swamped with everything at the moment, but while I'm thinking of it I want to link this to #314, which maybe we only alleviated slightly but didn't find the root cause of. My hunch is that (as last time) the gpu background calculation is still fine (and running async at the call which is taking forever) but the serial `clusterIO.get_file` calls might be slowing us down.

mProfiling things will surely be the fastest way to get after this, but I'm back-burner brainstorming what would cause `clusterIO.get_file` and `clusterIO.get_status` both to be performing poorly.

David-Baddeley, 2020-11-01:
There's a chance it could be network traffic inside the cluster - if we are saturating the network interfaces everything will slow down even if it's not super cpu intensive. Not aware of anything which should have changed this, but one thing which would rapidly bring stuff to a crawl is if computers were analysing non-local data for some reason. Easiest thing here is to check the task costs. 

Other things to check:

- total network traffic to each node (iftop?)
- number of remote sessions etc ... open on each node

 I'm pretty sure that there is an unused GBE port on each of the computers that make up the cluster you are using at the moment. It should, in theory, be possible to double the bandwidth by hooking them all up to the switch as well. Wouldn't jump into it just yet - at least not without being absolutely sure that network capacity is the problem, as there will likely be a bit of config mess to pair the connections etc ... . One of my medium term plans had been to channel separate the communications - put the dataserver traffic over one interface and everything else (rules, clusterUI, etc ...) over a second one.

barentine, 2020-11-02:
This is a little bold to claim 'eureka' already, but I'm willing to bet it's a bytes/str error in clusterIO caching. `_listSingleDir` encodes the url to bytes before using it as a key for `_dirCache`, but in `locate_file` we have a str (for sure) filename and directory and we check the `_dirCache` with that, which would be a miss on py3. 

`_locateCache`, however keys a str, and `_fileCache` keys whatever it's passed which in `ClusterPZFDataSource` is for sure a str.

barentine, 2020-11-02:
Sorry to have ignored your comment David. I appreciate the brainstorming but was writing up the other at the same time. I'll have a look with some logging statements to confirm #562 tomorrow, but I think the lack of caching on directory listing would be the level of doom we're experiencing, while all the other changes we've made since last April seem pretty small (e.g. adding an endpoint on the dataserver). You are right that we could team the connections from the nodes. I'm pretty sure our switch is 24 port so we'd have enough there too, and honestly that might just be a nice thing to do anyway for the cost, but since we've had this all working before at similar (13% higher now) bandwidth, and we've streamlined things since then (no more dcimg files, more calculations on the GPU / no redundant unit convert/unconverts), I'm optimistic we can get it working again with the same network bandwidth. We'll see!

barentine, 2020-11-02:
OK, so no-go on the str/bytes caching as mentioned in closing comment of #562. 

Question though about `_fileCache` which I'd like to look into is the units of the file size threshold of `1000000`. Looks like it should be in bytes in which case we'd be more than fine for the ~145 kB pzf files. 

Another cache related thing to check: now that we are actually using the serverfilter, are we ignoring it anywhere? We use the serverfilter in the key for `_locateCache`, meaning that we need to cache/look-for the file with or without the serverfilter on assignment and get. Logs in #314 show some URLs with the 'villains' serverfilter, but taskWorkerHTTP keyerrors without it

barentine, 2020-11-04:
discussion with David:
- [x] benchmark get cluster freespace on cluster of one (clusterIO.get_status returns in 4 - 10 ms from uncached on David's mac)
- [ ] mprofile clusterIO.py, HTTPDataServer.py, buffers.py, ClusterPZFDataSource.py
- [ ] double check the set of fresh background indices buffers.py on each frame, logger.debug()

note that clusterIO.listdir on a 30k pzf series is taking something like 100 ms (which we hit every datasource creation) -r1 -n1, and ~33 ms per if we run in a loop (should be cached)

barentine, 2020-11-24:
OK, finally some mprofiles. 

Pretty confused by this gem
[image]

But otherwise some interesting things:
- `get_ns().get_advertised_services` is still pretty atrociously slow:  ~200 ms even after subtracting a 10 s handicap for first get_ns.
- GPU fitting and actual buffer calculation are not bad (GPU fit for this data was ~27 ms per frame)
- fit result packing takes longer than I wish it did, but not our biggest issue at the moment
- making a 256 x 1024 pixel frame contiguous in memory takes about 3 ms (see detector.py)
- the last comment means that getSlice in the buffer updates are taking ~170 ms each

[20201124mprof-greenback.zip](https://github.com/python-microscopy/python-microscopy/files/5590662/20201124mprof-greenback.zip)

**Conclusion:**
our `get_advertised_services` performance, even on the zeroconf nameserver is shot, and that is almost certainly because as of 6 months ago we started checking if ports are open using `PYME.misc.sqlite_ns.is_port_open` on each call, which opens and closes a socket for each service...

@David-Baddeley what would you be comfortable with here? Can we add a dict cache on whether a port is open with a 1 hr expiration time? In practice we shouldn't need the check at all if we're unregistering services on their way down and no one is SIGKILLing or segfaulting...?

David-Baddeley, 2020-11-24:
I can definitely see how `is_port_open` could be problematic on every call, but de-registration is not foolproof (neither with zeroconf nor sqlite - you are especially likely to get problems on a crowded network) so we need to do something. Adding some form of caching here is probably the right thing to do, although the timeout should probably be significantly less than 1 hr.

David-Baddeley, 2020-11-24:
One potential solution would be to have a helper thread in the zeroconf listener which checked if the ports were open and removed the services if not (rather than have this in the `get_advertised_services()` call. Happy to give this a quick go.

David-Baddeley, 2020-11-24:
have a prototype in #645, but currently failing tests.

barentine, 2020-11-25:
Hi @David-Baddeley, looks like #645 did it. `get_advertised_services` is back to negligible runtime
[20201125mprof-greenback.zip](https://github.com/python-microscopy/python-microscopy/files/5597517/20201125mprof-greenback.zip)

I'd love to close this ticket, but get_status is still ~0.45 s per call. Note that the first run I'd killed the nodeservers but left the dataservers up, second one I restarted all the cluster processes first
```
In [1]: import sys

In [2]: sys.executable
Out[2]: 'C:\\Users\\Bergamot\\Anaconda2\\envs\\chaos\\python.exe'

In [3]: import timeit

In [4]: timeit.timeit('get_status("villains"); time.sleep(1)', setup='import time; from PYME.IO.clusterIO import get_st
   ...: atus', number=10)
Out[4]: 22.3657134

In [5]: timeit.timeit('get_status("villains"); time.sleep(1)', setup='import time; from PYME.IO.clusterIO import get_st
   ...: atus', number=10)
Out[5]: 14.294125399999999
```

I'll suggest I go ahead and write `get_polled_status` somewhere, though other than not clusterIO I don't remember where we'd discussed it living

closed 2020-12-16 by barentine