python-microscopy/python-microscopy #549: [BUG] dropping frames, i.e. PYMEAcquire / HTTPDataServer performance issues

barentine, 2020-10-28:
**Describe the bug**
Looking at data collected by a user
[image]

Final frame is 29999, only 27013 present. This was not the first series as acquired through the action manager - all of them have roughly the same number of frames

<details> <summary> associated metadata </summary> 

```
{
  "ActiveCamera": "HamamatsuORCA",
  "Camera.ADOffset": 101.753685,
  "Camera.Binning": [
    1,
    1
  ],
  "Camera.CycleTime": 0.0012570225563909773,
  "Camera.DarkMapID": "PYME-CLUSTER://villains/CALIBRATION/720795/dark_1ms.tif",
  "Camera.EMGain": 1,
  "Camera.ElectronsPerCount": 0.416613,
  "Camera.IntegrationTime": 0.0012570225563909773,
  "Camera.Model": "C11440-22C",
  "Camera.Name": "Hamamatsu ORCA Flash 4.0",
  "Camera.NoiseFactor": 1,
  "Camera.ROI": [
    180,
    896,
    1940,
    1152
  ],
  "Camera.ROIHeight": 256,
  "Camera.ROIOriginX": 180,
  "Camera.ROIOriginY": 896,
  "Camera.ROIWidth": 1760,
  "Camera.ReadNoise": 0.997,
  "Camera.SensorHeight": 2048,
  "Camera.SensorWidth": 2048,
  "Camera.SerialNumber": "720795",
  "Camera.StartCCDTemp": 0,
  "Camera.TrueEMGain": 1,
  "Camera.VarianceMapID": "PYME-CLUSTER://villains/CALIBRATION/720795/variance_1ms.tif",
  "Camera.Views": [
    0,
    3
  ],
  "CameraOrientation.FlipX": false,
  "CameraOrientation.FlipY": false,
  "CameraOrientation.Rotate": false,
  "EndTime": 1603834166.0494494,
  "Lasers.MPB560.AOTFSetting": 0.0,
  "Lasers.MPB560.LaserPower": 200,
  "Lasers.MPB560.MaxPower": 650.1,
  "Lasers.MPB560.MinPower": 0.0,
  "Lasers.MPB560.On": false,
  "Lasers.MPB560.Power": 0.0,
  "Lasers.MPB642.AOTFSetting": 22.0,
  "Lasers.MPB642.LaserPower": 1954.3973941368079,
  "Lasers.MPB642.MaxPower": 675.4,
  "Lasers.MPB642.MinPower": 0.0,
  "Lasers.MPB642.On": true,
  "Lasers.MPB642.Power": 600.0,
  "Lasers.OBIS405.AOTFSetting": 18.4,
  "Lasers.OBIS405.LaserPower": 100.0,
  "Lasers.OBIS405.MaxPower": 1.1,
  "Lasers.OBIS405.MinPower": 0.0,
  "Lasers.OBIS405.On": false,
  "Lasers.OBIS405.Power": 1.0,
  "Lasers.OBIS488.AOTFSetting": 18.9,
  "Lasers.OBIS488.LaserPower": 20.689655172413794,
  "Lasers.OBIS488.MaxPower": 5.316666666666666,
[... 267 lines cut ...]
zc.lockfile               2.0                        py_0
zeroc-ice36-python        3.6.5            py36h74a9793_6    ome
zeroconf                  0.26.3                   pypi_0    pypi
zipp                      3.3.0                      py_0
zlib                      1.2.11               h62dcd97_4
zstd                      1.4.5                h04227a9_0
```

</details>

**Additional context**

<details> <summary> instrument computer config </summary>

```
laser-slider-scaling: "linear"
dataserver-filter: "villains"
aotf-calibration-file: "C:\\Users\\Bergamot\\PYMEData\\CALIBRATION\\laser-intensities-20200604.json"
httpspooler-chunksize: 100
actionmanagerserver-address: "10.150.7.131"
clusterIO-hybridns: False
```

</details>
possibly related to: #545 ?

barentine, 2020-10-28:
dataserver logs look pretty tame

<details>

```
PYMEDataServer, running on python 3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 19:07:31) 
[GCC 7.3.0]

Serving from directory: /data/PYMEData/
Serving HTTP on 10.150.7.5 port 15348 ...
glob: pattern = Bergamot/2020_10_22/000/reg*
10.150.7.131 - - [26/Oct/2020 18:32:46] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:46] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:46] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:46] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:32:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:10] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:10] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:10] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:11] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:11] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:11] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:11] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 18:33:12] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:46] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:47] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:48] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:48] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:48] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:48] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:59] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:28:59] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:29:00] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:29:00] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:29:00] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]

10.150.7.131 - - [26/Oct/2020 21:29:01] code 404, message File not found - /Bergamot/2020_10_26/, [/data/PYMEData/Bergamot/2020_10_26/]
[... 1183 lines cut ...]
10.150.7.131 - - [28/Oct/2020 01:39:07] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/A4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/A4/001/]

10.150.7.131 - - [28/Oct/2020 01:39:08] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/A4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/A4/001/]

10.150.7.131 - - [28/Oct/2020 01:39:10] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/A4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/A4/001/]

10.150.7.131 - - [28/Oct/2020 01:57:02] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

10.150.7.131 - - [28/Oct/2020 01:57:03] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

10.150.7.131 - - [28/Oct/2020 01:57:04] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

[... cut ...]
10.150.7.131 - - [28/Oct/2020 01:57:21] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

10.150.7.131 - - [28/Oct/2020 01:57:22] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

10.150.7.131 - - [28/Oct/2020 01:57:24] code 404, message File not found - /Bergamot/2020_10_27/EMC07_02_S1/B4/001/, [/data/PYMEData/Bergamot/2020_10_27/EMC07_02_S1/B4/001/]

```

</details>

barentine, 2020-10-28:
Well, turns out a node died and had to reboot. Data is all still there, full 30k frames!

closed 2020-10-28 by barentine

David-Baddeley, 2020-10-28:
Maybe we should find a way of flagging it a bit more obviously when a node dies ...

barentine, 2020-10-28:
wouldn't hurt, but would suggest that as being low in priority relative to other open issues. Running PYME as a systemctl service might also be nice, and while that wouldn't necessarily point out when a node is down, a random reboot or so would at least bring PYME back online