python-microscopy/python-microscopy #551: [BUG]

David-Baddeley, 2020-10-29:
**Describe the bug**
Development install fails on most recent anaconda (py36 and py37), Win10

**To Reproduce**
- install miniconda
```
conda config --add channels david_baddeley
conda create -n PYME python=3.6 pyme-depends
conda activate PYME
cd python-microscopy
python setup.py develop

dh5view --test3d
```
Fails on pytables import. This would seem to be a bug in either the pytables or blosc packages.

**Workaround**
install `blosc` <=1.19

`conda install blosc=1.19.0`

**long term solution**
- pin blosc and/or pytables???

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3.6 & 3.7

David-Baddeley, 2020-10-29:
NB: will probably also effect non dev installs.

David-Baddeley, 2020-11-21:
Updated recipe for pyme-depends and pyme-depends pin in python-microscopy recipe (packages not built yet)

David-Baddeley, 2020-11-27:
Should be fixed, modulo testing with the latest packages and installers
 @zacsimile - Could you test when you have a chance?

zacsimile, 2020-12-04:
This was fixed as of the 20.11.25 packages. Tested on Win 10 and macOS 10.15.7.

closed 2021-01-28 by David-Baddeley