python-microscopy/python-microscopy #558: "Test this frame" is not working

lukasfue, 2020-10-30:
**Describe the bug**
"test this frame" gives me a traceback. I have only tested this with dcimg files.

**To Reproduce**
Steps to reproduce the behavior:
1. Open dh5view
2. Open dcimg file of raw data
3. attempt to "test this frame"
4. See error:

Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 931, in testFrame
    d = ft.ofd.filteredData.T
AttributeError: 'fitTask' object has no attribute 'ofd'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 288, in <lambda>
    bTestF.Bind(wx.EVT_BUTTON, lambda e : self.lmanal.OnTestFrame(e))
  File "c:\users\laf62\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 643, in OnTestFrame
    ft, fr = self.testFrame()
  File "c:\users\laf62\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 968, in testFrame
    if isinstance(ft.bg, np.ndarray):
AttributeError: 'fitTask' object has no attribute 'bg'

**Expected behavior**
A test of the current frame with the given settings

python-microscopy=20.10.05
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

**Additional context**
Add any other context about the problem here.

barentine, 2020-10-30:
can you include a screenshot of those settings @lukasfue? I'm wondering if we're trying to make a background buffer out of zero frames - is this on the first frame of the series?

lukasfue, 2020-10-30:
yeah I literally change nothing from the defaults
[image]

barentine, 2020-10-30:
Cool, I know what to fix. Thanks @lukasfue 

@David-Baddeley this looks like a result of my #341 short-circuiting before we default `bg` to zero

David-Baddeley, 2020-11-01:
The other option (which might be preferable, as the current fix is going to make testFrame give results which are not representative of what is going to happen when you click go) is to warn / pop up an error dialog if we test a frame we're not going to analyse. 

For a little historical context:
- You don't tend to run into this error on data spooled from PYMEAcquire as it will automatically set your position to the 'start of the actual data' when you open the file. 
- I always tend to move to a frame somewhere in the middle of the sequence before testing (of the middle of the bit of the sequence that has spooled by the time I get there) , as this tends to give the best indication of what the majority of your data will look like (absent all the artifacts which are often present in the first few frames)
- There used to be a `testFrames()` function which automatically tested 3 frames - near the start (but after startAt), in the middle, and at the end of the series. I think the code is still there, but not accessible through the GUI having fallen out of use once we switched to doing most analysis on the fly.

The above context because I think the problem is more a UI problem than a logic problem:
- testing a frame before `startAt` *should* probably fail, but should give a useful feedback as to why it fails
- the current UI behaviour is such that the most obvious/intuitive course of action (open a file, test) is not the ideal course of action (open a file, change to a frame away from the initial bleaching etc ... that is going to be representative, test)  

I'm not really sure what we should be doing here ... the automagically selecting 3 frames to test stuff seems a bit overkill (and is fragile once you start doing things like real-time analysis), opening a series so that the current frame is mid-way through the file also seems a bit counter-intuitive. Popping up an error dialog if the current frame is < startAt seems the easiest (if a little annoying).

barentine, 2020-11-01:
we can just do both - pop a dialog that says 'StartAt setting would exclude this frame, would you like to test this frame anyway?' and then just do #559 if they say 'test' and nothing if they say 'cancel'

barentine, 2020-11-02:
also just a bump that #546 is still an issue I have to remember when I'm clusterofone-ing at home, and could be silently tripping-up newcomers

David-Baddeley, 2020-11-24:
should now be fixed

closed 2020-11-24 by David-Baddeley