python-microscopy/python-microscopy #560: [BUG] rules.RecipeRule._task_template and HTTPRulePusher.RecipeRule._taskTemplate sets up for ujson failure

barentine, 2020-10-30:
**Describe the bug**
one extra \n no matter what, with a second available if you don't provide output_dir
```
{"id": "{{ruleID}}~0",
              "type": "recipe",
              "inputs" : "task_inputs",
              "taskdefRef" : "pyme-cluster://villains/RECIPES/calibrate-multiview-shifts.yaml",
              
              }

```
gives 
```
'{"id": "{{ruleID}}~0",\n              "type": "recipe",\n              "inputs" : "task_inputs",\n              "taskdefRef" : "pyme-cluster://villains/RECIPES/calibrate-multiview-shifts.yaml",\n              \n              }'
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/smeagol/miniconda3/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 352, in _poll_tasks
    self._update_tasks()
  File "/home/smeagol/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 221, in _update_tasks
    taskID, cost = rater.get_local()
  File "/home/smeagol/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 119, in get_local
    taskID, cost = self.next()
  File "/home/smeagol/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 113, in next
    return self.__next__()
  File "/home/smeagol/code/python-microscopy/PYME/cluster/rulenodeserver.py", line 82, in __next__
    task = json.loads(filled_template)
ValueError: Unexpected character in found when decoding object value

```
at least on later ujson versions

```
In [11]: task = '{"id": "{{ruleID}}~0",\n              "type": "recipe",\n              "inputs" : "task_inputs",\n              "taskdefRef" : "pyme-cluster://villains/RECIPES/
    ...: calibrate-multiview-shifts.yaml",\n              \n              }'

In [12]: task
Out[12]: '{"id": "{{ruleID}}~0",\n              "type": "recipe",\n              "inputs" : "task_inputs",\n              "taskdefRef" : "pyme-cluster://villains/RECIPES/calibrate-multiview-shifts.yaml",\n              \n              }'

In [13]: json.loads(task)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-13-d2cf0ddc3f63> in <module>
----> 1 json.loads(task)

ValueError: Unexpected character in found when decoding object value

In [14]: task = '{"id": "{{ruleID}}~0",\n              "type": "recipe",\n              "inputs" : "task_inputs",\n              "taskdefRef" : "pyme-cluster://villains/RECIPES/
    ...: calibrate-multiview-shifts.yaml",\n}'

In [15]: json.loads(task)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-15-d2cf0ddc3f63> in <module>
----> 1 json.loads(task)

ValueError: Unexpected character in found when decoding object value

In [16]: task = '{"id": "{{ruleID}}~0",\n              "type": "recipe",\n              "inputs" : "task_inputs",\n              "taskdefRef" : "pyme-cluster://villains/RECIPES/
    ...: calibrate-multiview-shifts.yaml"}'

In [17]: json.loads(task)
Out[17]: 
{'id': '{{ruleID}}~0',
 'type': 'recipe',
 'inputs': 'task_inputs',
 'taskdefRef': 'pyme-cluster://villains/RECIPES/calibrate-multiview-shifts.yaml'}
```
relevant to conversations in #485

David-Baddeley, 2020-11-02:
I think the extra character causing issues is a  comma (`,`), not a newline - ujson ignores extra newlines, but if `output_dir` is not specified in the recipe, we get a trailing comma in the json dict (e.g. `{"key1":val1, "key2":val2,}`) which breaks syntax.

barentine, 2020-11-02:
should be fixed with #561 - thanks David

closed 2020-11-02 by barentine