python-microscopy/python-microscopy #564: [BUG] coalescing averages (and does not sum) nPhotons

csoeller, 2020-11-02:
**Describe the bug**

Coalescing currently averages the `nPhotons` property, rather than summing it. Admittedly, only some fitfactories will produce this property, Gaussian 2D prominently does I believe.

**To Reproduce**
Steps to reproduce the behavior:
1. Open series with 2D Gaussian fitted events in VisGUI
2. Do `Corrections>Chaining` etc to coalesce events
3. check `nPhotons` property if present, pre- and post coalescing
4. Be disappointed...

**Expected behavior**

It seems to me that the `nPhotons` property needs special casing to sum if present.

**Screenshots**

[image]

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS
 - Python version: 2.7
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: NA

**Additional context**
Code from latest githgub as of writing.

barentine, 2020-11-02:
Thanks for catching that @csoeller. If you're in a rush and need to crunch something you can give #565 a shot

csoeller, 2020-11-02:
Excellent, thanks @barentine for that very quick response. Only became apparent when discussing histograms for a PhD thesis today! I suppose we can close as soon as the #565 is merged.

csoeller, 2020-11-03:
Fixed with #565.

closed 2020-11-03 by csoeller