python-microscopy/python-microscopy #566: [BUG] 'basedir' nor 'input_dir' always defined in ModuleCollection contexts

barentine, 2020-11-02:
**Describe the bug**
output.py docs state we should always have `basedir` and `filestub` in the context, however batchProcess does not include `basedir`, and neither does anything run on the cluster (taskWorkerHTTP tries to add `input_dir` if we have an input called `input`). Additionally, basedir + filestub doesn't include the file extension, so if you want to do something like upload the principle h5r file to omero you'd better know it's an h5r file. Finally, docs state `filestub` rather than `file_stub`...

**Expected behavior**
choose `input_dir` or something like it to be available

barentine, 2020-11-02:
note that looking for an input named `input` is also less useful now that we pull anything out of hdf5 files and name them their tablenames. In other words, should we look for `FitResults` or `input`? 

Could be nice if `ModuleCollection` input loading/checking added to the context dictionary, so we had `FitResults~file_stub`, `astigmatism_calibration~file_stub`, etc.. When you write the recipe you have the namespace keys right in front of you, so would be nice and straightforward/flexible.

David-Baddeley, 2020-11-02:
In the first instance, this is a documentation error in the output modules. The two context variables which are usually present are `file_stub` and `output_dir` (although `output_dir` is not 100% guaranteed). `taskWorkerHTTP` also defines `input_dir`, although one should be super careful about using this (having things implicitly depend on the input directory is a good way to end up with unexpected behaviour). I'm not sure why `input_dir` made it in in the first place, but in an ideal world it probably shouldn't be in the context for output modules.

I'm not super keen on having per-input file stubs etc ... as this opens the possibility for far to much logic and confusion in the output modules. I think it makes more sense to essentially keep with `output_dir` , 'file_stub` pair of parameters, keeping the logic of what should be the `file_stub` separate. We definitely need to do something about the case where there is not an input called `"input"`, but it's probably best to find a way of inferring (or better yet, specifying) which is the primary input. In the yaml representation there is a clear 0th input, but as it stands that probably won't be preserved over load/save cycles. 

Reading between the lines, this came up when trying to get omero uploads to work well. Medium/long term it might make more sense to do uploads outside of the recipe framework as they are not a natural fit (they are effectively 100% side-effect and need to infer lots of stuff which falls into the "recipe modules shouldn't need to know this" box).

barentine, 2020-11-02:
I can certainly work with `file_stub` and `output_dir` being the only context variables one can count on. I think it does point to maybe a bit of a gap in our metadata though - I find it odd to have stuff floating in a recipe namespace and not be able to link it back to the URI where it came from.