python-microscopy/python-microscopy #567: PYMEVis Error: cannot open .h5f after updating PYME

HelenSun09, 2020-11-02:
Cannot open . h5r file in PYME vis after updating PYME using git pull https://github.com/python-microscopy/python-microscopy
dh5view is still running perfectly.

Error whilst running File>&Open
==============================================
python-microscopy=20.10.05
python=2.7.16, platform=win32
numpy=1.15.4, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\public\pyme\python-microscopy\PYME\ui\progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "c:\users\public\pyme\python-microscopy\PYME\LMVis\visCore.py", line 402, in OnOpenFile
    self.OpenFile(filename)
  File "c:\users\public\pyme\python-microscopy\PYME\LMVis\visCore.py", line 741, in OpenFile
    self.recipeView._layout()
  File "c:\users\public\pyme\python-microscopy\PYME\recipes\vertical_recipe_display.py", line 122, in _layout
    pan = node.edit_traits(parent=item, kind=kind, view='pipeline_view_min')
  File "C:\Users\Public\miniconda2\lib\site-packages\traits\has_traits.py", line 2017, in edit_traits
    view = self.trait_view(view)
  File "c:\users\public\pyme\python-microscopy\PYME\recipes\base.py", line 238, in trait_view
    import traitsui.api as tui
  File "C:\Users\Public\miniconda2\lib\site-packages\traitsui\api.py", line 25, in <module>
    from .editor import Editor
  File "C:\Users\Public\miniconda2\lib\site-packages\traitsui\editor.py", line 46, in <module>
    from .undo import UndoItem
  File "C:\Users\Public\miniconda2\lib\site-packages\traitsui\undo.py", line 22, in <module>
    import collections.abc
ImportError: No module named abc

David-Baddeley, 2020-11-02:
ugh - could you include the results of `conda list` (looks like it might be a traits / traitsui version issue)

David-Baddeley, 2020-11-02:
actually, that's super-super odd, as just git-pulling shouldn't have done anything to the other packages installed (and this would clearly seem to be traits related).

csoeller, 2020-11-02:
Could it possibly be that a `conda update` or something similar was done as well at that or a similar time? I have on occasion been left with broken installs after trying to update conda packages.

HelenSun09, 2020-11-02:
traits                    5.2.0            py27h0c8e037_0
traitsui                  7.1.0                      py_0

alabaster                 0.7.12                   py27_0
asn1crypto                1.4.0                      py_0
babel                     2.8.0                      py_0
backports                 1.0                        py_2
backports.functools_lru_cache 1.6.1                      py_0
backports_abc             0.5                        py_1
blas                      1.0                         mkl
ca-certificates           2020.10.14                    0
cairo                     1.14.10           vc9hf2422aa_4  [vc9]  anaconda
certifi                   2020.6.20          pyhd3eb1b0_3
cffi                      1.13.2           py27hcfb25f9_0
chardet                   3.0.4                 py27_1003
cheroot                   5.8.3            py27h6ad7a9d_0
cherrypy                  11.0.0           py27h8c72c6f_0
colorama                  0.4.4                      py_0
conda                     4.8.3                    py27_0
conda-package-handling    1.6.0            py27h3cc03e0_0
console_shortcut          0.1.1                         3
cryptography              2.8              py27hcfb25f9_0
cycler                    0.10.0           py27h59acbbf_0
cython                    0.29.14          py27hc56fc5f_0
decorator                 4.4.2                      py_0
dispatch                  1.0                      py27_0    david_baddeley
docutils                  0.15.2                   py27_0
dyekinetics               2.0                       dev_0    <develop>
enum-compat               0.0.2                    py27_0    david_baddeley
enum34                    1.1.2                    py27_0    david_baddeley
fftw                      3.3.4                         0    david_baddeley
freetype                  2.10.0               h3bc688b_1    conda-forge
functools32               3.2.3.2                    py_3    conda-forge
future                    0.18.2                   py27_0
futures                   3.3.0                    py27_0
git                       2.23.0               h6bb4b03_0    anaconda
hdbscan                   0.8.24           py27h83e69c8_0    conda-forge
hdf5                      1.10.1            vc9h9d13b45_0  [vc9]  anaconda
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2              vc9h3e79e4e_0  [vc9]  anaconda
idna                      2.10                       py_0
imagesize                 1.2.0                      py_0
intel-openmp              2019.4                      245
ipaddress                 1.0.23                     py_0
jinja2                    2.11.2                     py_0
joblib                    0.14.1                     py_0    conda-forge
kiwisolver                1.1.0            py27hdc96acc_0    conda-forge
libblas                   3.8.0           7_h8933c1f_netlib    conda-forge
libcblas                  3.8.0           7_h8933c1f_netlib    conda-forge
liblapack                 3.8.0           7_h8933c1f_netlib    conda-forge
libpng                    1.6.37               h7a46e7a_0
m2w64-gcc-libgfortran     5.3.0                         6
m2w64-gcc-libs            5.3.0                         7
m2w64-gcc-libs-core       5.3.0                         7
m2w64-gmp                 6.1.0                         2
m2w64-libwinpthread-git   5.0.0.4634.697f757               2
markupsafe                1.1.1            py27h0c8e037_0
matplotlib                2.2.5                         1    conda-forge
matplotlib-base           2.2.5            py27h6595424_1    conda-forge
menuinst                  1.4.16           py27h0c8e037_0
mkl                       2018.0.3                      1
mpld3                     v0.3                     py27_0    david_baddeley
msys2-conda-epoch         20160418                      1
netifaces                 0.10.9                   py27_0    david_baddeley
networkx                  2.2                      py27_1
numexpr                   2.6.8            py27h286ddaf_0
numpy                     1.15.4          py27hde85f9e_1002    conda-forge
openssl                   1.1.1d               h0c8e037_3
packaging                 20.4                       py_0
pandas                    0.23.4           py27h39f3610_0
[... 32 lines cut ...]
singledispatch            3.4.0.3                 py_1001
six                       1.10.0                   py27_0    david_baddeley
snowballstemmer           2.0.0                      py_0
sphinx                    1.8.5                    py27_0
sphinxcontrib             1.0                      py27_1
sphinxcontrib-websupport  1.2.4                      py_0
sqlite                    3.30.1               h0c8e037_0
tempora                   1.11                     py27_0
toposort                  1.5                      py27_3    david_baddeley
tornado                   5.1.1           py27h0c8e037_1000    conda-forge
tqdm                      4.50.2                     py_0
traits                    5.2.0            py27h0c8e037_0
traitsui                  7.1.0                      py_0
typing                    3.7.4.1                  py27_0
ujson                     1.35             py27h0c8e037_0
urllib3                   1.25.7                   py27_0
vc                        9                    h7299396_1
vs2008_runtime            9.00.30729.1         hfaea7d5_1
wheel                     0.35.1                     py_0
win_inet_pton             1.1.0                    py27_0
wincertstore              0.2              py27hf04cefb_0
wxpython                  4.0.4            py27hc56fc5f_0
yaml                      0.1.7             vc9h88dcfe9_1  [vc9]  anaconda
zeroconf                  0.18.0                   pypi_0    pypi
zlib                      1.2.11            vc9ha80f4e8_1  [vc9]  anaconda

csoeller, 2020-11-02:
I notice traitsui is on v 7.1.0 which was only released 13 days ago (so you must have upgraded this recently). traitsui v7.x states "**this release drops support for Python 2**" which IMHO is the explanation as this is run in a python 2.7 environment. traistui needs to be downgraded to 6.X, I currently run 6.1.3 in a 2.7 environment.

HelenSun09, 2020-11-02:
Thanks a lot! How would I achieve that? I tried the "conda install --revision 7", but did not seem to work.

zacsimile, 2020-11-02:
`conda install traitsui=6` should do it

zacsimile, 2020-11-02:
Do we need to add a pin for py2?

HelenSun09, 2020-11-03:
conda install traitsui=6 solved the problem.

closed 2020-11-03 by HelenSun09

David-Baddeley, 2020-11-03:
@zacsimile - before we add a pin, it might be worth a bit more investigation .... traitsui=7 for py27 should have never made it into the channels. I had initially suspected conda-forge as a culprit, but it looks to be the core anaconda channels in this case. Anaconda seem to be getting really slack about their curation these days. It would be nice to think that they've just stopped caring about py27 and that things would get better one we're on 3.x, but there's also the blosc and pytables version problem I have an open issue about and that affects py3.x as well.

zacsimile, 2020-11-03:
Yeah, I was a bit surprised `conda` allowed this. Maybe we just give up on pinning for py2, but it sounds like we'll keep needing to be a little careful on py3.

barentine, 2020-11-03:
[image]