python-microscopy/python-microscopy #569: [ENHANCEMENT] Add a "recenter" button to PYMEVis

zacsimile, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
It's easy to lose track of your data in the 3D viewer in PYMEVis. Normally I address this by hitting "C". Folks who don't know this shortcut are screwed.

**Describe the solution you'd like**
I'd like to add a button between the 3D axis shifters and LUT button called "Recenter" that is the equivalent of pressing "C".

**Describe alternatives you've considered**
- Double-clicking the window recenters the data.

**Are you prepared to work on it?**
Yes

David-Baddeley, 2020-11-03:
A couple of quick questions to consider:

- what should it be called / what icon? (A "home" icon would be consistent with, e.g. matplotlib)
- also add a menu item, along with `View-->Fit` etc ...?
- do we also need buttons for View -->Fit etc ??? 
- what should our 'home' button do? Just centre? centre and fit? centre, fit and reset rotation? something else?

zacsimile, 2020-11-03:
- To me a home icon resets everything, i.e. we go back to 2D fit with no rotation. 
- I find "C" to be more helpful than `View->Fit` or a home icon, but I'm not sure how others feel. I'd advocate for an icon other than home, maybe a target instead, and for this being the default option. The reason I like "C" is it keeps my rotation and zoom and makes it easier to continue rotating/translating/zooming from the point I'm already at. A home button would force me to redo everything.
- We should definitely have a menu item for "C". I vote "View->Recenter".
- I never use `View->Fit`, I always zoom instead. Again, that might just  be me. That said, a `View->Fit` shortcut key sounds like a great idea. How about "f"?

David-Baddeley, 2020-11-03:
I tend to use `View-->Fit` a lot more than `c`, but ultimately it will depend on what type of data you are looking at.

zacsimile, 2020-11-04:
We could make the target button run `View->Fit`, that's fine.

csoeller, 2020-11-04:
When I was maintaining a separate branch for our work in Exeter (now a long time ago) I always had a shortcut for View>Fit and found it mega convenient. I had not been aware of the C shortcut. View>Fit is super-useful when you reset yourself. I need to ask, is that the same as FitROI? I tended to use a lot after selecting a small detail with F8.

zacsimile, 2020-11-04:
New proposal (sorta same as the old proposal):

- Home button (view reset) next to the plane buttons
- Move shortcuts to meta keys and put them in the View Menu
- `View -> Home` is `Alt-H` (equivalent of `Alt-C` and `Alt-R`)
- `View -> Fit` is `Alt-F`
- `View -> Recenter` is `Alt-C`
- `View -> Reset rotation` is `Alt-R`
- `View -> FitROI` is `Alt-8` (corresponds to `F8`) (Re: @csoeller, FitROI fits the ROI to the viewing window while View->Fit will fit whatever is selected, including an ROI, to within the view bounds, see images below)
- `View -> Stereo` is `Alt-S`
- `View -> Reduce stereo distance` is `Alt-[`
- `View -> Increase stereo distance` is `Alt-]`

Fit:
[image]

Fit ROI:

[image]

closed 2020-11-23 by David-Baddeley