python-microscopy/python-microscopy #570: [ENHANCEMENT] Change zoom speed on PYMEVis canvas.

zacsimile, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
Even slight scrolling, especially on track pads, causes drastic changes in the zoom on some machines.

**Describe the solution you'd like**
Allow the user to adjust the zoom speed (change `ZOOM_FACTOR` in `PYME.LMVis.gl_render3D`). Ideally this setting would be saved when the user adjust the speed and be re-loaded at opening of PYMEVis.

I might put this under a new dialog, `File>Settings`. Alternatively, holding control while scrolling could adjust the zoom speed, but this might be a bit weird.

**Describe alternatives you've considered**
Leave it hardcoded and slow down zooming for everyone.

**Are you prepared to work on it?**
Yes

David-Baddeley, 2020-11-03:
This is going to be a bit hard to get right/intuitive and might be worth doing a little more research on. It would be great if we knew which hardware/os the complaint came from (or if it is observable on multiple machines). I wonder if the issues might arise from, a pretty edge use case e.g. running a windows VM on a mac). As well as having `ZOOM_FACTOR` set-able, having sensible defaults would also be good. When I initially determined the current values, I tested win with wheel mouse, win with trackpad, linux, and OSX on a few different machines.  A good first step would be to make a `PYME.config` option for it which would over-ride the defaults. Whether we put a GUI round it or just document how to set the config option should probably depend on how common an issue it is.

The rolls-royce solution would be to record the time between scroll events and if it's super short, pop up a dialog prompting the user to change the zoom factor. Otherwise it's always going to be a setting that will take some effort to discover and it's going to feel wrong in the meantime.

zacsimile, 2020-11-04:
I did a bit of research. As best I can tell, our zoom speed is strictly set by the rate at which `wx` throws `EVT_MOUSEWHEEL`. I can't find any data on how this rate is set. I know that it does not change if I change the speed of my trackpad (macOS 10.15.7), so it's not tied to system mouse settings in the way I thought it would be. Most video games come with a mouse sensitivity settings, and while I've found detailed posts on what the best setting is for "realistic" sniping on Counter Strike and other games, there does not seem to be a consistent pattern to the math. It entirely varies on how we set up aspect ratios/zoom factors in our own GL window. As such, I think a slider that adjusts `ZOOM_FACTOR` between maybe `0.1` and `2` is the most intuitive solution. We can empirically set defaults for Mac/Windows (as we have done), but they're likely to need a bit of tweaking from system to system and user to user as best I can tell.

David-Baddeley, 2020-11-04:
By research, I really meant trying a number of different systems to work out how common is it for the defaults to be badly wrong. IE is this something that effects <1% of users, in which case it could probably be dealt with a config option and a line in the docs, but no GUI, ~5-10% would suggest that we implement some form of GUI, ~30% or higher would suggest we should get a bit better at guessing sensible default values.

barentine, 2020-11-04:
The zoom speed on my macbook trackpad is pretty untenable for me, nor do I think I've ever seen someone trackpad it without some difficulty/frustration. I tend to think most of our users don't submit UI feedback since we have a speed enough learning curve anyway :upside_down_face:  so hearing anything probably warrants a good check-in if not action (probably exactly what @David-Baddeley 's accounting for with those low percentages).

zacsimile, 2020-11-17:
Confirmed that the zoom speed on Windows 10 using a track pad is way too fast. Each swipe changes the zoom factor by 2 (as you'd expect from the code). I recommend setting a default of 1.1x for all operating systems and adding a slider to adjust the feel.

closed 2020-12-04 by zacsimile