python-microscopy/python-microscopy #571: [ENHANCEMENT] Update number of localizations displayed after a filter in the PYMEVis Data pipeline

zacsimile, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
Right now, if I look at `Data pipeline` before and after applying a filter I have
[image]
and
[image]
where the number of events listed under `filtered_localizations` is the same. Visually I can see this is not the case.

**Describe the solution you'd like**
Have the data pipeline accurately reflect the number of localizations present after applying a filter.

**Are you prepared to work on it?**
Yes

David-Baddeley, 2020-11-03:
This has been on my annoyance list for ages. A bit messy to implement which is why it's not been done yet (have sort of been punting it until I can put the whole lot in html). Some options:

- completely recreate the `vertical_recipe_display` on any change to the pipeline (undesirable from a performance perspective)
- only update the event counts when the pipeline updates (need to keep track of the static text elements when creating the panel - maybe in a dict - and find some way of triggering off any recipe change. Not as bad as completely recreating the panel, but still involves doing a `len()` on everything
 - only update the label for outputs which have changed. Probably ideal, but hard to implement.

barentine, 2020-11-18:
this set @zacsimile?

zacsimile, 2020-11-18:
Yeah

closed 2020-11-18 by zacsimile