python-microscopy/python-microscopy #579: [PROPOSAL] make ImagePyramid choose TileIO dynamically

adrianhollow, 2020-11-04:
**Is your feature request related to a problem? Please describe.**
The TileIO in ImagePyramid is currently hardcoded for PZFTileIO. If a user wants to load an ImagePyramid from another file format, the code in \_\_init\_\_ needs to be adapted accordingly.

**Describe the solution you'd like to implement**
Set the TileIO based on file extensions in base_dir to avoid changing the code manually.

**What potential side effects do the proposed changes have?**
- additional code will run in ImagePyramid.\_\_init\_\_ (looping through files in base_dir until a file extension is recognized)
- new TileIO classes will need to be registered in a dict or won't be recognized automatically
- if no file extension is recognized, an Exception will be raised (and end the program)

David-Baddeley, 2020-11-04:
Hi Adrian, it would be good to better understand the use case here. Are you trying to load image pyramids created in different software? Or do you just want to experiment with the different backends?

As its currently written `ImagePyramid` is designed, in the first instance, for creating an image pyramid (although it can also open an existing one that it has created). It's not really a general-purpose image pyramid class at this point (which is not to say that we couldn't make it one). For the creation use case there obviously won't be any files present to infer type from. Both for this reason, and for performance, it might be preferable to just specify the backend as an argument to the constructor. We could also write the backend used into the metadata and read this when re-opening a pyramid if a backend is not specified.

David-Baddeley, 2020-11-04:
More generally, `ImagePyramid` is due a bit of an overhaul (and a good bit of documentation) re the distinction between loading and creating.  It was written quickly and is very much alpha-quality code. My first thought would be to leave the constructor as the default way of creating a new pyramid and add a `load_exiting(...)` class method for the loading case.

We'd still need to change the constructor a bit as it over-writes a bunch of parameters which it should read out of the metadata when loading an existing pyramid (although we could probably handle most of this by reading those parameters in `load_exisiting()` and passing them to the constructor).

adrianhollow, 2020-11-04:
Hi David, I am trying to open different datasets, some of which have been saved as pzf-files and others as npy-files. When I run "dh5view SUPERTILE:\<path\>" at the moment, I can only open datasets whose corresponding TileIO is hardcoded in \_\_init\_\_ of ImagePyramid. I thought it would make sense to adjust the TileIO selection at runtime so a user can avoid touching the code when they want to open a different dataset.

I admit I didn't think of pyramid creation and the reason why the TileIO and file format are preconfigured is clear to me now. Using a constructor parameter or the metadata in my use case sounds more stable than inference from file endings and I would be happy to change that in the code.

If I try to open a dataset and there is no corresponding entry in the metadata, would it make more sense to raise an error, assume a default file format or try type inference?

closed 2021-02-23 by adrianhollow